realsenseai/librealsense #7823: Error occured when connecting D455i in realsense-viewer. Invalid value in rs2_get_option.

gongyeted, 2020-11-20:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | 05.12.08.200 | 
| Operating System & Version |  Ubuntu 16 | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC |
| SDK Version                            |  { legacy / 2.38.1 }                          | 
| Language                            |  {C++ }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
erroe occurred when using realsense viewer saying "Invalid value in rs2_get_option xxx". It happens every time I open "Stereo Module" .

MartyG-RealSense, 2020-11-20:
Hi @gongyeted   If the camera is displayed in the RealSense Viewer's side-panel, are you able to use the Stereo Module if you go to the **More** option and select the menu option **Hardware Reset** to perform a reset of the camera, please?

[image]

MartyG-RealSense, 2020-11-26:
Hi @gongyeted   Do you require further assistance with this case, please?   Thanks!

gongyeted, 2020-11-30:
> Hi @gongyeted Do you require further assistance with this case, please? Thanks!

I tried to use the option Hardware Reset but the error still happens when I using the Stereo Module. It seems that the realsense-viewer is still working despite this error occurs.

gongyeted, 2020-11-30:
> Hi @gongyeted Do you require further assistance with this case, please? Thanks!

yes I need further assistance, please. The detailed error information is:
 "Timestamp: 1606726107010.314941
Severity: Error
Description: Invalid Value in rs2_get_option(options:0x76ba4e0, option:Emitter On Off):
hwmon command 0x7d( 0 0 0 0 ) failed (response -21= No data to return)"

MartyG-RealSense, 2020-11-30:
If the Viewer can work normally after the error occurs, it may be a timing hiccup when changing an option that the Viewer then automatically recovers from until the next time that the option is used.

What method did you use to build librealsense on Ubuntu, please: was it DKMS packages, building from source code, or using RSUSB?

MartyG-RealSense, 2020-12-07:
Hi @gongyeted  Have you solved your problem, please?

closed 2020-12-21 by MartyG-RealSense