realsenseai/librealsense #7825: Trying to play rosbag with pyrealsense but get the error: Invalid file format, file does not contain topic "/file_version

IcookTacos, 2020-11-20:
Hello! I've recorded a rosbag using the intel realsense depth camera D455. I'm playing the rosbag in one terminal on loop and trying to initialize a pyrealsense2 pipeline but I get the following error:

**RunTimeError: Failed to resolve request. Request to enable_device_from_file("testBag.bag") was invalid, Reason: Failed to create ros reader: Invalid file fromat, file does not contain topic "/file_version" nor "/FILE_VERSION"**

Here's a code snippit from where the error occours:

```
import rospy
import cv2
import pyrealsense2 as rs
from sensor_msg.msg import Image
from cv_bridge import CvBridge

def process_image(msg):
    pipeline = rs.pipeline()
    cfg = rs.config()
    cfg.enable_device_from_file("testBag.bag")
    profile = pipeline.start(cfg)
```

Thanks in advance for any help I can get!

MartyG-RealSense, 2020-11-20:
Hi @IcookTacos   If you recorded the rosbag in a ROS terminal then it may not be entirely compatible with being read by the RealSense SDK (librealsense).  The reasons for this are described in the discussion in the link below, which is about a RealSense Python user who experienced the same error.

https://github.com/IntelRealSense/librealsense/issues/3020

IcookTacos, 2020-11-21:
Thanks for the response!

closed 2020-11-21 by IcookTacos