realsenseai/librealsense #7826: Installation Failure: patching file drivers/media/usb/uvc/uvcvideo.h Hunk #1 FAILED at 153.

jbourne15, 2020-11-20:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   |  | 
| Operating System & Version |   Ubuntu 16.04  | 
| Kernel Version (Linux Only)    |  4.15.0-107-generic  | 
| Platform                                 | arm64 |
| SDK Version                            |  2                          | 
| Language                            |  C                          | 

Trying to install into a docker image with the base image: https://hub.docker.com/r/arm64v8/ros/. After following the installation instructions found at https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md it failed to patch the uvcvideo module.

Running the following command: 

    ./scripts/patch-realsense-ubuntu-lts.sh

Output:

patching file drivers/media/usb/uvc/Makefile
patching file drivers/media/usb/uvc/uvc_driver.c
Hunk #1 succeeded at 208 with fuzz 2 (offset 5 lines).
patching file drivers/media/usb/uvc/uvcvideo.h
Hunk #1 FAILED at 153.
1 out of 1 hunk FAILED -- saving rejects to file drivers/media/usb/uvc/uvcvideo.h.rej
patching file drivers/media/v4l2-core/v4l2-ioctl.c
Hunk #1 succeeded at 1248 (offset 2 lines).
patching file include/uapi/linux/videodev2.h
Hunk #1 succeeded at 680 (offset 12 lines).

jbourne15, 2020-11-23:
I ended up installing through https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md and configuring kernel drivers.

closed 2020-11-23 by jbourne15

jindanChina, 2020-12-31:
How do you solve this problem? I've been puzzling this problem for a long time.