realsenseai/librealsense #7827: T265 Enable Mapping?

rlabs-oss, 2020-11-20:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { T265 } | 
| Firmware Version                   | (0.2.0.951) | 
| Operating System & Version |   {Linux (Ubuntu 20.04)  | 
| Kernel Version (Linux Only)    |  ( 5.4.0-54)                                         | 
| Platform                                 | PC and PI4  |
| SDK Version                            |  { 2.39.0 }                          | 
| Language                            |  {C++ }                          | 
| Segment			|  {Mad science}                          | 

### Issue Description

For the [pose example](https://github.com/IntelRealSense/librealsense/blob/master/examples/pose/rs-pose.cpp)
Is mapping enabled as default? If not please provide example code how to set the "RS2_OPTION_ENABLE_MAPPING" option. 

Regardless of indoor or outdoor setting: pose_data.tracker_confidence and pose_data.mapper_confidence are always returning a tracking confidence of 2 and mapping confidence of 0.  (low and failed).

Thank you.

mindThomas, 2020-11-22:
You need to modify the corresponding `rosparam` before launching the nodelet.
See https://github.com/mindThomas/JetsonCar/blob/master/ros/jetsoncar_sensors/launch/t265.launch#L16

rlabs-oss, 2020-11-22:
@mindThomas not using ROS, but will check the ROS driver to see how that parameter is parsed and set. Thank you.

closed 2020-11-22 by rlabs-oss

mindThomas, 2020-11-22:
Oh sorry, I didn't realize or thought about that.
With the SDK you need to get the sensor from the device first. How about trying this out: https://github.com/mindThomas/JetsonCar/blob/master/lib/cpp/realsense/realsense.cpp#L71-L72