realsenseai/librealsense #7828: How to download my own code to the VPU of T265

wangce888, 2020-11-21:


dorodnic, 2020-11-24:
This feature is not supported for T265

eric-schleicher, 2020-11-24:
Step 1: but luxonis DepthAI.

Step 2: wait for micropython support

-- 
---------------------------------
CONFIDENTIALITY NOTICE: The contents of this email message and any
attachments are intended solely for the addressee(s) and may contain
confidential and/or privileged information and may be legally protected
from disclosure. If you are not the intended recipient of this message or
their agent, or if this message has been addressed to you in error, please
immediately alert the sender by reply email and then delete this message
and any attachments. If you are not the intended recipient, you are hereby
notified that any use, dissemination, copying, or storage of this message
or its attachments is strictly prohibited.

closed 2026-10-23 by MartyG-RealSense