realsenseai/librealsense #7829: Make install error "undefined reference to `gluPerspective'" in Ubuntu 20.04 LTS

prantoran, 2020-11-21:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435i | 
| Firmware Version                   | - | 
| Operating System & Version |    Linux (Ubuntu 20.04 LTS)  | 
| Kernel Version (Linux Only)    |  5.4.0-53-generic  | 
| Platform                                 | PC |
| SDK Version                            | ~ 2.40.0 | 
| Language                            |  C/C++ | 
| Segment			|  VR/AR  | 

### Issue Description

I was able to pass through the [missing config.h](https://github.com/IntelRealSense/librealsense/issues/7822) error and how I got this `undefined reference to `gluPerspective'` error while building the examples:

```
...
[ 58%] Building CXX object src/gl/CMakeFiles/realsense2-gl.dir/align-gl.cpp.o
In file included from /usr/include/c++/9/backward/strstream:50,
                 from /home/prantoran/work/src/github.com/IntelRealSense/librealsense/src/gl/align-gl.cpp:21:
/usr/include/c++/9/backward/backward_warning.h:32:2: warning: #warning This file includes at least one deprecated or antiquated header which may be removed without further notice at a future date. Please use a non-deprecated interface with equivalent functionality instead. For a listing of replacement headers and interfaces, consult the file backward_warning.h. To disable this warning use -Wno-deprecated. [-Wcpp]
   32 | #warning \
      |  ^~~~~~~
[ 59%] Building C object src/gl/CMakeFiles/realsense2-gl.dir/__/__/third-party/realsense-file/lz4/lz4.c.o
[ 59%] Building CXX object src/gl/CMakeFiles/realsense2-gl.dir/__/__/third-party/easyloggingpp/src/easylogging++.cc.o
[ 59%] Linking CXX shared library librealsense2-gl.so
[ 59%] Built target realsense2-gl
[ 59%] Building CXX object examples/hello-realsense/CMakeFiles/rs-hello-realsense.dir/rs-hello-realsense.cpp.o
[ 60%] Linking CXX executable rs-hello-realsense
[ 60%] Built target rs-hello-realsense
[ 60%] Building CXX object examples/software-device/CMakeFiles/rs-software-device.dir/rs-software-device.cpp.o
[ 60%] Linking CXX executable rs-software-device
/usr/bin/ld: CMakeFiles/rs-software-device.dir/rs-software-device.cpp.o: in function `draw_pointcloud(float, float, glfw_state&, rs2::points&)':
/home/prantoran/work/src/github.com/IntelRealSense/librealsense/examples/software-device/../example.hpp:773: undefined reference to `gluPerspective'
/usr/bin/ld: /home/prantoran/work/src/github.com/IntelRealSense/librealsense/examples/software-device/../example.hpp:777: undefined reference to `gluLookAt'
/usr/bin/ld: CMakeFiles/rs-software-device.dir/rs-software-device.cpp.o: in function `draw_pointcloud_wrt_world(float, float, glfw_state&, rs2::points&, rs2_pose&, float*, std::vector<rs2_vector, std::allocator<rs2_vector> >&)':
/home/prantoran/work/src/github.com/IntelRealSense/librealsense/examples/software-device/../example.hpp:839: undefined reference to `gluPerspective'
collect2: error: ld returned 1 exit status
make[2]: *** [examples/software-device/CMakeFiles/rs-software-device.dir/build.make:90: examples/software-device/rs-software-device] Error 1
make[1]: *** [CMakeFiles/Makefile2:1293: examples/software-device/CMakeFiles/rs-software-device.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
```

I tried the following:

- Installed other packages mentioned in the [instruction](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md#prerequisites):
```
 sudo apt-get install git libssl-dev libusb-1.0-0-dev pkg-config libgtk-3-dev 
```

- Installed freeglut3 but it did not fix the problem:
```
sudo apt-get install freeglut3 freeglut3-dev
```

- I executed the following for `glfw` [support](https://github.com/IntelRealSense/librealsense/issues/7360#issuecomment-693525921)
```
sudo apt-get install libglfw3-dev libgl1-mesa-dev libglu1-mesa-dev at 
```
Btw, the D435i camera works plug and play with Cheese webcam software and shows grayscale image with IR points, that is so cool.

prantoran, 2020-11-21:
Okay, restart and cmake rebuild solved it for some reason

closed 2020-11-21 by prantoran

sebastienL-cog, 2021-12-01:
Got the same error ; had to add "find_package(glfw3)" line 75 in file librealsense/CMakeLists.txt

Here is my new CMakeLists.txt :+1: 
```
# global project configuration
# macro definition located at "CMake/global_config.cmake"
global_target_config()

include(CMake/install_config.cmake)

# HERE***
find_package(glfw3)

add_subdirectory(wrappers)
if ( ( BUILD_EXAMPLES OR BUILD_PC_STITCHING ) AND BUILD_GLSL_EXTENSIONS )
    find_package(glfw3 3.3 QUIET)
    if(NOT TARGET glfw)
        message(STATUS "GLFW 3.3 not found; using internal version")
        set(GLFW_INSTALL ON CACHE BOOL "" FORCE)
        add_subdirectory(third-party/glfw)
    endif()
    add_subdirectory(src/gl)
endif()
```