realsenseai/librealsense #7830: Best way to get depth from aligbed_depth_to_color topic?

IcookTacos, 2020-11-21:
Hello! I'm trying to solve a problem using a intel realsense D455. Right now I have a rosbag captured from a car running by a stop-sign. I'm trying to find the stop-sign and the depth from the camera to the sign. I have a node that recognizes the stop sign and gives me the pixels on the screen where the sign is. So my issue now is that I want to take that pixel-data of x and y and from the aligned_depth_to_color image find the depth to that pixel.

So let's say my stop sign is at the positions x and y. When I run this code for example:

```
depth_image = bridge.imgmsg_to_cv2(ros_image,desired_encoding='passthrough')
depth_array = np.array(depth_image,dtype=np.float32)
print(depth_array[x,y])
```

The problem is that I get around 2300 printed all the time.

Thanks for any advice you could give me!

MartyG-RealSense, 2020-11-22:
Hi @IcookTacos   The link below may provide some useful insights.   It discusses a method of using Python to retrieve depth information from the **/aligned_depth_to_color**topic of a bag recorded in ROS.

https://github.com/IntelRealSense/realsense-ros/issues/1312

IcookTacos, 2020-11-22:
@MartyG-RealSense Amazing! Thanks for the help. Will look into that thread!

MartyG-RealSense, 2020-11-29:
Hi @IcookTacos   Do you still require assistance with this case, please?   Thanks!

IcookTacos, 2020-12-01:
No I think I have solved it. Thanks for the help!

closed 2020-12-01 by IcookTacos

MartyG-RealSense, 2020-12-01:
Great to hear - thanks for the update!