realsenseai/librealsense #7831: Sometimes I get the abnormal cloud data but the function return is OK in 2.38 x64_linux_16.04 ubuntu

shiguangzqz, 2020-11-22:
I'm doing a long machine test，but sometimes ,the cloud data get from the camera is abnormal.
I try close and  then reopen the camera,it still return the abnormal data .There was no movement of the machine during the entire test.
However ，It will return to normal data after 10 minutes(may be shorter time) when I am not using 。
Show my code :
void getDataFromCamera()
{
//to avoid to memory leak，we use the static
static rs2::points points;
static rs2::pointcloud pc;

       auto frameset = _pipe.wait_for_frames();
        rs2::align align(rs2_stream::RS2_STREAM_COLOR);
        auto aligned_frameset = align.process(frameset);
        auto color = aligned_frameset.get_color_frame();     
        auto depth = aligned_frameset.get_depth_frame();
        //rs2::points points;    
        //rs2::pointcloud pc;
        pc.map_to(color);     
        points = pc.calculate(depth);   
//do  something with the points.data
 }

MartyG-RealSense, 2020-11-22:
Hi @seebq   If the camera returns to normal after a period of time such as ten minutes or less when you are not using the camera, I would recommend checking first whether it is a temperature related event.  If the camera's projector temperature exceeds 60 degrees C for several seconds then the firmware driver's safety mechanism should halve the laser power to try to bring the temperature down .  If the temperature should continue to be excessive after that then the laser is shut off. 

Reducing the laser power makes the depth image more sparse, so I can foresee how you may get abonormal data if the laser power was halved by an excessive temperature event.  You can incorporate temperature reading code into your application.  An easy test though would be to simply touch the camera casing whilst the camera is active when the problem occurs and see whether the casing is hot to the touch.

shiguangzqz, 2020-11-23:
Thanks!But in which file can I refer to the code for getting camera temperature?

MartyG-RealSense, 2020-11-23:
If you are using C++, I hope that the discussion in the link below will be useful to you for getting the ASIC and Projector temperatures.

https://github.com/IntelRealSense/librealsense/issues/6735

shiguangzqz, 2020-11-24:
hello ，i try this code learn form the question #6735 to get the temperature :
    auto sensors=ctx.query_all_sensors();
    logInfoDataData("the sensors size is "+ to_string(sensors.size()));
    logInfoDataData("start to read the temperature");
    for(auto sensor : sensors)
    {
       if(sensor.supports(RS2_OPTION_PROJECTOR_TEMPERATURE))
        {
            auto temp=sensor.get_option(RS2_OPTION_PROJECTOR_TEMPERATURE);
            logInfoDataData("the TEMPERATURE is "+to_string(temp));
        } 
        else 
            logInfoDataData("the sensor is not support the temperature");
    }

However ,the log  show :
[2020-11-&d-15-23-17-556]:the sensors size is 3
[2020-11-&d-15-23-17-556]:start to read the temperature
[2020-11-&d-15-23-17-556]:the sensor is not support the temperature

I try to reduced the frequency of polling temperature(per 10 second to ask for once ),but it still not return the temperature 
I am using the Intel RealSense Depth Camera D435 ,the Firmware version is 5.12.08.200,Min FW Version is 5.12.05

MartyG-RealSense, 2020-11-24:
Reading camera temperatures with code can be an awkward experience, unfortuntately.  I did find another example of projector temperature code that you could try.

https://github.com/IntelRealSense/librealsense/issues/866#issuecomment-357461253

shiguangzqz, 2020-11-25:
OK。I can get the temperature from the #866 question. Thanks !

closed 2020-11-25 by shiguangzqz

MartyG-RealSense, 2020-11-25:
Great news - thanks very much for the update!