realsenseai/librealsense #7834: Found 0 realsense devices

nyanmn, 2020-11-23:
I have installed [legacy version of realsense](https://github.com/IntelRealSense/librealsense/tree/v1.12.1).
Lsusb gives 

Bus 002 Device 002: ID 05e3:0626 Genesys Logic, Inc. 
**Bus 002 Device 018: ID 8086:0ad0 Intel Corp. 
Bus 002 Device 019: ID 8086:0acb Intel Corp.** 
Bus 002 Device 017: ID 0451:8025 Texas Instruments, Inc. 
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 03f0:1198 Hewlett-Packard HID-compliant mouse
Bus 001 Device 003: ID 046d:c31c Logitech, Inc. Keyboard K120
Bus 001 Device 002: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 001 Device 008: ID 0451:8027 Texas Instruments, Inc. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

So device is detected.

modinfo uvcvideo | grep "version:" gives
**version:        1.1.2.realsense-1.3.12
srcversion:     BE1821F94C3D410077ACD10**

demsg delivers

```
[  427.064823] uvcvideo 2-2.1:1.4: Entity type for entity Camera 6 was not initialized!
[  427.065391] input: Intel RealSense 3D Camera ZR300 as /devices/pci0000:00/0000:00:15.0/usb2/2-2/2-2.1/2-2.1:1.4/input/input13
[ 1066.061282] usb 1-2: USB disconnect, device number 7
[ 1066.207835] usb 2-2: USB disconnect, device number 13
[ 1066.207840] usb 2-2.1: USB disconnect, device number 15
[ 1066.260084] usb 2-2.2: USB disconnect, device number 14
[ 1591.199645] perf: interrupt took too long (2545 > 2500), lowering kernel.perf_event_max_sample_rate to 78500
[ 1961.632031] perf: interrupt took too long (3193 > 3181), lowering kernel.perf_event_max_sample_rate to 62500
[ 2393.224409] perf: interrupt took too long (4019 > 3991), lowering kernel.perf_event_max_sample_rate to 49750
[ 3095.836639] perf: interrupt took too long (5049 > 5023), lowering kernel.perf_event_max_sample_rate to 39500
[ 4005.999918] perf: interrupt took too long (6331 > 6311), lowering kernel.perf_event_max_sample_rate to 31500
[ 5471.587427] usb 2-2: new SuperSpeed USB device number 17 using xhci_hcd
[ 5471.607615] usb 2-2: New USB device found, idVendor=0451, idProduct=8025
[ 5471.607621] usb 2-2: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[ 5471.608868] hub 2-2:1.0: USB hub found
[ 5471.608905] hub 2-2:1.0: 2 ports detected
[ 5471.735155] usb 1-2: new high-speed USB device number 8 using xhci_hcd
[ 5471.883708] usb 1-2: New USB device found, idVendor=0451, idProduct=8027
[ 5471.883717] usb 1-2: New USB device strings: Mfr=0, Product=0, SerialNumber=1
[ 5471.883721] usb 1-2: SerialNumber: 860E0079A8EF
[ 5471.884616] hub 1-2:1.0: USB hub found
[ 5471.884660] hub 1-2:1.0: 2 ports detected
[ 5471.963344] usb 2-2.2: new SuperSpeed USB device number 18 using xhci_hcd
[ 5471.984509] usb 2-2.2: LPM exit latency is zeroed, disabling LPM.
[ 5471.985410] usb 2-2.2: New USB device found, idVendor=8086, idProduct=0ad0
[ 5471.985419] usb 2-2.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 5471.985424] usb 2-2.2: Product: CX3-UVC
[ 5471.985428] usb 2-2.2: Manufacturer: Cypress
[ 5471.987008] uvcvideo: Found UVC 1.10 device CX3-UVC (8086:0ad0)
[ 5471.988309] uvcvideo 2-2.2:1.0: Entity type for entity Extension 3 was not initialized!
[ 5471.988318] uvcvideo 2-2.2:1.0: Entity type for entity Processing 2 was not initialized!
[ 5471.988323] uvcvideo 2-2.2:1.0: Entity type for entity Camera 1 was not initialized!
[ 5472.539725] usb 2-2.1: new SuperSpeed USB device number 19 using xhci_hcd
[ 5472.564398] usb 2-2.1: New USB device found, idVendor=8086, idProduct=0acb
[ 5472.564407] usb 2-2.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 5472.564412] usb 2-2.1: Product: Intel RealSense 3D Camera ZR300
[ 5472.564416] usb 2-2.1: Manufacturer: Intel Corp
[ 5472.564420] usb 2-2.1: SerialNumber: SN_3511805355
[ 5472.567773] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera ZR300 (8086:0acb)
[ 5472.571316] uvcvideo 2-2.1:1.0: Entity type for entity Extension 2 was not initialized!
[ 5472.571323] uvcvideo 2-2.1:1.0: Entity type for entity Camera 1 was not initialized!
[ 5472.572891] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera ZR300 (8086:0acb)
[ 5472.576069] uvcvideo: Unable to create debugfs 2-19 directory.
[ 5472.576401] uvcvideo 2-2.1:1.2: Entity type for entity Camera 4 was not initialized!
[ 5472.577797] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera ZR300 (8086:0acb)
[ 5472.581085] uvcvideo: Unable to create debugfs 2-19 directory.
[ 5472.581400] uvcvideo 2-2.1:1.4: Entity type for entity Processing 7 was not initialized!
[ 5472.581405] uvcvideo 2-2.1:1.4: Entity type for entity Camera 6 was not initialized!
[ 5472.581722] input: Intel RealSense 3D Camera ZR300 as /devices/pci0000:00/0000:00:15.0/usb2/2-2/2-2.1/2-2.1:1.4/input/input14

```

But when I run realsense-viewer

Got error as 

**13:23:47 [Info] /home/upsquared/librealsense/src/context.cpp:381 - Found 0 RealSense devices (mask 254)**

What could be wrong?

MartyG-RealSense, 2020-11-23:
Hi @nyanmn   The **RealSense Viewer** program is for RealSense SDK 2.0 (librealsense2), which does not have ZR300 compatibility.  

In addition to Legacy Librealsense, there was also a dedicated official SDK produced by Intel for the ZR300 called **RealSense SDK For Linux** that added a large amount of additional features on top of Legacy Librealsense.  I do not know if this SDK still functions but I will provide the link to its GitHib so that you can look at it.

https://github.com/IntelRealSense/realsense_sdk_zr300

MartyG-RealSense, 2020-11-29:
Hi @nyanmn   Do you still require assistance with this case, please?   Thanks!

nyanmn, 2020-11-30:
Hi I'll try later. Now we can close. Thanks

closed 2020-11-30 by MartyG-RealSense