realsenseai/librealsense #7838: Unable to turn IR projector on

hcostelha, 2020-11-23:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

MartyG-RealSense, 2020-11-23:
Hi @hcostelha    I tested with a D435i in SDKs 2.40.0 and 2.39.0 with the default settings that you used but was unable to replicate your problem.

If it is a new camera, could you go to the **More** option at the top of the RealSense Viewer's options side-panel and select the **Install Recommended Firmware** menu option to install in your camera the firmware driver recommended for SDK 2.40.0 please?

[image]

If you still have no IR pattern after the firmware update, can you confirm whether you have an IR pattern on the images if you move the camera close to your hand, please?

[image]

hcostelha, 2020-11-23:
Hi,

Thanks for the reply. I updated the camera as you suggested, but the same firmware was installed (I alraedy updated the camera firmware the first time I connected the device with RealSense Viewer). Regarding viewing the pattern, I tested again and, as shown below, still no pattern is shown. Any more ideas and/or suggestions?
[image]

MartyG-RealSense, 2020-11-23:
Next, let's try manually changing the Laser Power value to see if it can toggle the pattern on.  Move the Laser Power slider to its minimum (which hides the dot pattern) and then click on the middle of the slider to return it to its default value of '150'.  Doing so should toggle the pattern from invisible to visible under normal circumstances.

[image]

[image]

dorodnic, 2020-11-24:
@hcostelha - any unusual log messages in Viewer output box? The firmware may decide to disable laser, if it finds something wrong with it (most extreme example at high temperatures to prevent from overheating) This would usually come with some kind of error message

hcostelha, 2020-11-25:
Thanks again for the reply. I already had tried changing the laser power and, doing as you described, produced no change whatsoever.
I have also checked the log and, around the time I change the laser power,I get the messages below, but I think they (the error ones) are not related with the laser, and they just appear when I am applying changes to the sensor parameters.
```
08:53:12 [Error] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\proc\synthetic-stream.cpp:48 - Exception was thrown during user processing callback: Error occured during execution of the processing block! See the log for more info
08:53:18 [Info]  - Setting Laser Power to 150
08:53:18 [Error] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\proc\synthetic-stream.cpp:48 - Exception was thrown during user processing callback: Out of frame resources!
```

MartyG-RealSense, 2020-11-25:
Hi again @hcostelha   I have performed further testing but have still been unable to replicate your problem with the same Viewer, SDK, firmware and camera configuration. 

@dorodnic  A thought occurs to me.  If the laser activation status is starting on laser setting 1 (Laser), I wonder if something could be causing the status on the system of @hcostelha  to shift one value up to '2' (Laser Auto).  When Laser Auto is set in the Viewer, the IR pattern is off.

[image]

hcostelha, 2020-11-25:
I have just tried starting with the Emitter Enabled set to "Off", and then changing it to "Laser", but not change happens. 
Do you have any further ideias, or should I assume this is a defective camera and I have to return it for a replacement or repair?

MartyG-RealSense, 2020-11-25:
@hcostelha   Let's wait for further input from @dorodnic as he is the RealSense SDK Manager.

MartyG-RealSense, 2020-12-03:
Hi @dorodnic  Could you provide input on the emitter problem above please?   Thanks!

https://github.com/IntelRealSense/librealsense/issues/7838#issuecomment-733902309

MartyG-RealSense, 2020-12-10:
Hi @hcostelha   Do you still have your problem with the IR emitter please?

hcostelha, 2020-12-10:
> Hi @hcostelha Do you still have your problem with the IR emitter please?

Yes, nothing changed since my last post.

MartyG-RealSense, 2020-12-11:
I researched the case again from the start but still could not replicate the non-appearance of the dot pattern that you are experiencing, unfortunately.  

On Windows, could you try downloading the standalone .exe version of the RealSense Viewer - **Intel.RealSense.Viewer.exe** - from the **Assets** list of 2.40.0 and launching it with a double-leftclick from the folder location that you downloaded it to (not with the desktop launch icon).

https://github.com/IntelRealSense/librealsense/releases/tag/v2.40.0

[image]

hcostelha, 2020-12-11:
Thanks for the reply. I have downloaded the RealSense Viewer only as suggested, tried the various approaches suggested earlier, but the result i still the same, I am unable to view the dot pattern.

MartyG-RealSense, 2020-12-12:
I ran further extensive tests but the only way I was able to replicate your dot-less image when the laser is enabled was to put my finger over the projector to block it.  Can you confirm please that the projector on your D435i is unobscured?

[image]

MartyG-RealSense, 2020-12-15:
It does look like a D435i if the image that you kindly supplied is brightened and flipped horizontally.

[image]

hcostelha, 2020-12-15:
> It does look like a D435i if the image that you kindly supplied is brightened and flipped horizontally.

Yes, the camera was flipped when I took the picture. I was not saying it was not the same camera is just that the colors in the inner circle of the projector are more homogeneous in your picture than in mine. Nevertheless, at least exteriorly, the sensor is not obscured.

MartyG-RealSense, 2020-12-16:
Hi @hcostelha  I have tested again but not been able to replicate the problem, unfortunately.  So I will seek advice from Intel.  Thanks for your patience!

MartyG-RealSense, 2020-12-16:
Hi again @hcostelha   Intel provided me with advice about your case.  Could you install the old **2.32.1** SDK using the Windows installer file in the **Assets** file list for that version and see whether the emitter functions in the Viewer that is installed by that version.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.32.1

[image]

hcostelha, 2020-12-17:
> Hi again @hcostelha Intel provided me with advice about your case. Could you install the old **2.32.1** SDK using the Windows installer file in the **Assets** file list for that version and see [...]

I have just tried, and I am still unable to view the patter, but this time I get a different behavior. When I open the viewer I get the error bellow.
[image]

Furthermore, If I set the option "Emitter On Off" to 1 I get the error above again.

MartyG-RealSense, 2020-12-17:
Okay, I will feed your latest result back to Intel.   Thanks again for your great patience.

MartyG-RealSense, 2020-12-17:
I have received further advice from Intel.

1.  They believe that the error message that you received from librealsense SDK 2.32.1 is fine, since it is an old SDK being used with a newer firmware.

2.   Do you have a different PC that you can try the camera on?  Or do you have another D435 that you can try on the same computer?

3.  Could you please install the old firmware version **5.10.6** on the camera and use it with the librealsense SDK version 2.32.1 that you have already installed?

[image]

If the advice above does not resolve your IR emitter problem then we can discuss the next step.

hcostelha, 2020-12-17:
> I have received further advice from Intel.
>
> 1. They believe that the error message that you received from librealsense SDK 2.32.1 is fine, since it is an old SDK being used with a newer firmware.
> 2. [...]

I have been using another RealSense camera, model D415, which works fine and I can see the pattern just fine. Unfortunately I do not have another D435i right now (thus my urgency on getting this one to work), so I cannot try with the same model.

I have tried the camera on another PC, which as Linux installed instead of Windows, and I get the same behavior: no infrared pattern (even after trying several options, as suggested previously).

> 3. Could you please install the old firmware version **5.10.6** on the camera and use it with the librealsense SDK version 2.32.1 that you have already installed?

I have installed firmware version **5.10.6**, and it always fails when rebooting the camera after the upgrade (downgrade). Furthermore, after the downgrade, I no longer am able to use the camera on USB 3 ports, only on USB 2 ports, so I am not sure what's going on here.
If I do the downgrade/upgrade to **5.10.6** with the device connected to an USB2 port, it finishes the upgrade procedure and restarts without any problem. However, I am unable to visualize the IR streams in this case, since it gives me the following error (even if I change the data type used):
[image]

In this case, with firmware **5.10.6**, I do not even get the option to start/configure the RGB camera:
[image]

MartyG-RealSense, 2020-12-18:
For the next step in this case, could you post a comment below with an email address that you can be privately contacted at, please?  The email address can be edited out of the comment once we have it.

hcostelha, 2020-12-18:
> For the next step in this case, could you post a comment below with a personal email address that you can be privately contacted at, please? The email address can be edited out of the comment once we have it.

You can reach me at hugocostelha@ipleiria.pt

MartyG-RealSense, 2020-12-18:
Ok, I have received it.   Thanks very much.   You can edit the address out now if you wish.  I will discuss the next step with Intel and we will communicate with you at this email address.

MartyG-RealSense, 2020-12-25:
Hi @hcostelha   As your case was continued by email, is it okay to close the case here?   This will not affect any ongoing email discussions with Intel that you are currently having.  Thanks!

hcostelha, 2020-12-25:
> Hi @hcostelha As your case was continued by email, is it okay to close the case here? This will not affect any ongoing email discussions with Intel that you are currently having. Thanks!

Shouldn't we confirm if it is an hardware problem, so that we can write here the conclusion for the given situation?

MartyG-RealSense, 2020-12-26:
If you wish to keep this case open until the email discussion has reached a conclusion and then make a final post on this GitHub discussion then that is fine.   I will leave it open for a further time period.

MartyG-RealSense, 2021-01-08:
Hi @hcostelha   Could you confirm pleae whether you wish to keep this case open until the email discussion has reached a conclusion and then make a final post on this GitHub discussion.   Otherwise this case will be closed in 7 days if we do not hear from you.   Thanks!

hcostelha, 2021-01-08:
> Hi @hcostelha Could you confirm pleae whether you wish to keep this case open until the email discussion has reached a conclusion and then make a final post on this GitHub discussion. Otherwise this case will be closed in 7 days if we do not hear from you. Thanks!

Hi @MartyG-RealSense , yes, I would prefer to have feedback about what was the camera problem and have it written here to close the issue.

MartyG-RealSense, 2021-01-08:
Okay, thanks.   If your problem was not solved by the email communication then please let me know and I will communicate this back to Intel so they can arrange further email contact with you.

canuck6ix, 2021-01-13:
I have similar issue. D435i and no IR pattern visible. In addition, Infrared streams are really dark. I did hardware reset, updated to the latest firmware, played with the laser power as suggested above, to no avail.

MartyG-RealSense, 2021-01-13:
Hi @canuck6ix  Could you post a comment below with a personal email address that you can be privately contacted at, please? The email address can be edited out of the comment once we have it.

canuck6ix, 2021-01-13:
> Hi @canuck6ix Could you post a comment below with a personal email address that you can be privately contacted at, please? The email address can be edited out of the comment once we have it.

Sure it's my handle at gmail.

MartyG-RealSense, 2021-01-13:
Ok thanks very much, I will pass it on to the appropriate Intel person so that you can be contacted by email about your issue.

canuck6ix, 2021-01-13:
> Ok thanks very much, I will pass it on to the appropriate Intel person so that you can be contacted by email about your issue.

Thanks!

RealSenseSupport, 2021-01-13:
Hi @canuck6ix 
We just received your personal email address. Thanks.
Can you let us know if you're currently using FW5.12.9 and RS Viewer 2.40.0?

Thanks

canuck6ix, 2021-01-13:
Hello thanks for your email.

Ubuntu 20.04
RS viewer 2.41.0
FW 5.12.10

RealSenseSupport, 2021-01-13:
Thanks. Did you get any error message when you tried to turn on the Laser?

Let's check if there is hardware error on laser projector: (see pic below)
Would you please open the command line window, and try command of "FLAGB_READ" in the command line? 
When you select to turn On "Laser", what do you get from the above command? does it show "Flag B: 00" or "Flag B: 01"? Thanks.
[image]

canuck6ix, 2021-01-13:
16:50:25 [Warn] /home/jenkins/workspace/LRS_SDK_CI_Debian/src/common/output-model.cpp:869 - Unrecognized command 'FLAGB_READ'
16:50:55 [Warn] /home/jenkins/workspace/LRS_SDK_CI_Debian/src/common/output-model.cpp:869 - Unrecognized command 'flagb_read'

RealSenseSupport, 2021-01-14:
@canuck6ix would you mind to try it on Windows 10? thanks

canuck6ix, 2021-01-14:
> @canuck6ix would you mind to try it on Windows 10? thanks

I don't have a Windows machine. I have a D435 of which the IR projector works fine. 
This D435i unit used to work fine and now has issues.

canuck6ix, 2021-01-14:
Here are some warning and error messages upon launching realsense-viewer:
````
$ realsense-viewer 
 14/01 12:01:25,671 WARNING [139808588670720] (ds5-factory.cpp:669) Cannot read RGB calibration table
 14/01 12:01:25,671 WARNING [139808588670720] (ds5-factory.cpp:784) invalid RGB extrinsic was identified, recovery routine was invoked
 14/01 12:01:25,722 ERROR [139808588670720] (ds5-factory.cpp:823) RGB Extrinsic recovery routine failed
 14/01 12:01:25,773 WARNING [139808588670720] (ds5-factory.cpp:669) Cannot read RGB calibration table
 14/01 12:01:25,893 WARNING [139808588670720] (backend-v4l2.cpp:1357) Pixel format 36315752-1a66-a242-9065-d01814a likely requires patch for fourcc code RW16!
 14/01 12:01:42,481 WARNING [139808323352320] (hdr-merge.cpp:38) HDR Merge filter cannot process frames because relevant metadata params are missing
 14/01 12:01:50,275 ERROR [139808331745024] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: Out of frame resources!
 14/01 12:01:50,308 ERROR [139808331745024] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: Out of frame resources!

RealSenseSupport, 2021-01-19:
@canuck6ix Hi I just received a defective D435i which also had laser projector issue. But that one doesn't have the problem to run the command of "flagb_read". You said you have another camera - D435? can you try same command on it?
By the way, your Viewer log shows that the RGB calibration table corrupted. Can you please check the calibration table by using [Dynamic Calibration Tool](https://downloadcenter.intel.com/download/29618?v=t&elq_cid=6395910_ts1611076742706&erpm_id=9509336_ts1611076742706)? please use the command of "_Intel.Realsense.CustomRW.exe_". (see the [User Guide](https://www.intel.com/content/www/us/en/support/articles/000026723/emerging-technologies/intel-realsense-technology.html))
Thanks

RealSenseSupport, 2021-01-26:
@canuck6ix Hi just wonder if you have any update? or anything else needed from us? Thanks

MartyG-RealSense, 2021-02-05:
Hi @hcostelha  Do you require further assistance with this case please?   Thanks!

@canuck6ix   Are you also still having problems?

hcostelha, 2021-02-05:
Hi @MartyG-RealSense, I am still waiting on a final statement from Intel, which currently has the defective(?) camera for assessment.

MartyG-RealSense, 2021-02-05:
Hi @hcostelha  Intel have been actively looking at your case.   I have passed on your request for a final statement about your case.  Thanks so much for your patience!

RealSenseSupport, 2021-02-09:
Hi @hcostelha The failure analysis may take a while. We will let you know the result. Can we close this issue now?

hcostelha, 2021-02-09:
> Hi @hcostelha The failure analysis may take a while. We will let you know the result. Can we close this issue now?

Dear @RealSenseSupport , if you think that it is not relevant to incude the result here, feel free to close the issue.

closed 2021-02-26 by RealSenseCustomerSupport