realsenseai/librealsense #7839: [Question] Getting a still image on demand - latency

prybicki, 2020-11-23:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.12.09.00 | 
| Operating System & Version | Ubuntu 20.04 (Linux Mint 20) | 
| Kernel Version (Linux Only)    |  5.4.0-54-generic | 
| Platform                                 | x86_64  |
| SDK Version                            |  2.40.0 | 
| Language                            |  C/C++ | 
| Segment			|  Robot | 

### Issue Description
My goal is to be able to get quickly (<<100 ms) a single frameset (still image) (RGB, depth, infras) at random moments from one of the many (>>10) connected RealSense cameras in highest resolution.

Because continuous streaming is not possible due to the USB3 bandwidth constraint, I tried following simple method:
``` C++
rs2::pipeline pipeline;
while (true) {
  pipeline.start();
  // measure time (start)

  rs2::frameset frames = pipeline.wait_for_frames();
   // measure time (capture)

  pipeline.stop();
  // measure time (stop)
}
```
However, I got the following results, which are too high and unpredictable for my application:
[image]

**Can anyone point me to what can be done to achieve my goal?**
I don't mind going into a lower-level API or even touching v4l2/uvc if that's necessary.

MartyG-RealSense, 2020-11-23:
Hi @prybicki   For a multiple camera application, such as the **rs-multicam** example program, it is recommended to use **poll_for_frames()** instead of wait_for_frames().  The differences between these instructions are described in the link below.

https://github.com/IntelRealSense/librealsense/issues/2422#issuecomment-423254709

With poll_for_frames() it is necessary to control when the CPU is put to sleep and for how long.

agrunnet, 2020-11-23:
@prybicki Have you considered using the genlock feature where you can trigger RealSense cameras to grab single (or multiple) frames at any time?

MartyG-RealSense, 2020-11-23:
@agrunnet   Thanks for the suggestion!  I thought about genlock too but @prybicki is using D415, which doesn't have the global shutter that genlock needs.

prybicki, 2020-11-23:
Thank you for your quick responses.
@MartyG-RealSense I tried using `while(!poll_for_frames(&frame));` and it seems only marginally faster & more stable. I think the issue here lies in starting & stopping the stream, I suspect that it requires power mode switches and/or exposure calibration which produces that delay when waiting/polling for the very first time. Is it possible to avoid it?

@agrunnet I read the following [whitepaper](https://dev.intelrealsense.com/docs/external-synchronization-of-intel-realsense-depth-cameras?_ga=2.197988015.1668201718.1606143333-772402966.1606143333) and it seems to be exactly what I need. As far as I understood, I have to set `RS2_OPTION_INTER_CAM_SYNC_MODE` before starting the stream, start the stream, send a 1.8V trigger on SYNC pin and the camera will *only* then send frames' data over the USB. Is that correct?
Also - I read that the GenLock feature is supported only by D435 and D435i. Is that still correct? Are there any new models to be released in the future?

MartyG-RealSense, 2020-11-23:
@prybicki   The new D455 model has global shutters too.   It is the first model to have these fast shutters on both the RGB and depth sensors.

dorodnic, 2020-11-24:
@prybicki - formally, this type of "snapshot" feature is not supported by RealSense cameras. However, from my experience, if you do not enqueue V4L2 buffers, uvcvideo will not request frames from the device, reducing bandwidth. Right now we are constantly dequeuing and enqueueing buffers in a dedicated thread, but you could try adding some sort of condition variable that you would control via C API that would block enqueue buffs. Important to note that this is not tested, I'm just sharing my thoughts.

prybicki, 2020-11-25:
@dorodnic Thank you, it sounds very interesting, I would like to try out this hack.

I had added a busy-wait `while(stream_paused);` before each invocation of ioctl(QBUF/DQBUF) in backend-v4l2.cpp and I toggled it whenever I wanted to stop/resume the stream. I was still getting frames via callback given to `rs2::sensor::start()`, but unfortunately I'm still observing traffic on the USB (via usbtop/usbmon), so I suspect that's a wrong way to go.

Could you please explain to me more precisely what would you do in backend-v4l2.cpp to make uvcvideo stop requesting frames over USB?

Also, I'd appreciate any other documentation (if there is any) on the structure of the backend code. It's quite a complex and threaded beast, so it would help me a lot.

MartyG-RealSense, 2020-12-03:
Hi @prybicki   Do you still need answers to your questions above, please?

Regarding your documentation request: there is an hour long YouTube seminar about the librealsense2 software architecture presented by @dorodnic that you may find useful.  

https://youtube.com/watch?v=uz2dGbjA3j4

prybicki, 2020-12-07:
@MartyG-RealSense Big thanks for linking the seminar!
I think for now I'll try some workaround like using a Raspberry Pi 4 per camera.
Closing the issue, thanks for your help.

closed 2020-12-07 by prybicki