realsenseai/librealsense #7841: Install Windows SDK 2016 1.2

AlbertoSabater, 2020-11-23:
Hello, I was attempted to install an old version of the librealsense SDK (2016, 1.2), with support for R200 and SR300 depth cameras and hand skeleton tracking. However, I cannot find anywhere the executable files for the Windows installation. How can I find them? Is there any other way to install them?

Thanks in advance!

MartyG-RealSense, 2020-11-23:
Hi @AlbertoSabater   The 2016 SDK for Windows was a completely separate SDK from librealsense and the two were incompatible.  During that time, the Windows SDK was the official Intel RealSense SDK and librealsense was an open-source alternative (which is now the sole official SDK as **librealsense2**).

If you plan to use the R200 then the **2016 R2** SDK that supports F200, R200 and SR300 should be used, as R200 support was dropped in the 2016 R3 version.

2016 R2 is a 1.8 gb installer file that contains all SDK functions in a single package.  R3 is a modular system where you install the **Essentials**  core package first and then optional modules for features suc as hand tracking and scanning.  I recommend opting for the all-in-one R2 version.

Please bear in mind that the R2 installer file has a slow download rate, so it is best to launch the download and then do something else for a while during its download.  Also, the 2016 SDK is no longer supported and has not been updated for some years, so success of installation cannot be guaranteed on modern versions of Windows 10.

You can obtain 2016 R2 and R3 from the page at the link below.  Scroll down the page to find the file lists.

https://software.intel.com/content/www/us/en/develop/articles/realsense-sdk-windows-eol.html

MartyG-RealSense, 2020-11-29:
Hi @AlbertoSabater   Do you still require assistance with this case please?   Thanks!

MartyG-RealSense, 2020-12-05:
Case closed due to no further comments received.

closed 2020-12-05 by MartyG-RealSense

NooshinNajafian, 2024-04-14:
Hi . I can not find the download link of appropriate sdk for r200 camera. please send me the link.nooshin.najafian@gmail.com ----thanks for your kindness

MartyG-RealSense, 2024-04-14:
Hi @NooshinNajafian  The **librealsense1** SDK at the link below is compatible with the R200 camera model.  I will also email the link to you.

https://github.com/IntelRealSense/librealsense/tree/v1.12.1