realsenseai/librealsense #7843: L515 only recognized as USB 2 device

ofl1, 2020-11-23:
|Required Info||
|-|-|
|Camera Model|L515|
|Serial Number|F0265069|
|Firmware Version|n/a|
|SDK Version|2.39.0|
|OS|Linux Manjaro|
|Kernel|5.9.3|
|Platform|PC|

### Issue Description

Hi all, I bought an L515 camera from the Intel store a week ago. It doesn't work. The RealSense Viewer always reports *Found 0 RealSense devices*. At the OS level, the USB 3 device always returns an error. After that, it will only be recognized as a USB 2 device. Output of dmesg:
```
[Mo Nov 23 14:32:24 2020] usb 6-1: new SuperSpeed Gen 1 USB device number 2 using xhci_hcd
[Mo Nov 23 14:32:25 2020] usb 6-1: unable to read config index 0 descriptor/all
[Mo Nov 23 14:32:25 2020] usb 6-1: can't read configurations, error -71
[Mo Nov 23 14:32:25 2020] usb 6-1: new SuperSpeed Gen 1 USB device number 3 using xhci_hcd
[Mo Nov 23 14:32:25 2020] usb 6-1: unable to read config index 0 descriptor/all
[Mo Nov 23 14:32:25 2020] usb 6-1: can't read configurations, error -71
[Mo Nov 23 14:32:25 2020] usb usb6-port1: attempt power cycle
[Mo Nov 23 14:32:25 2020] usb 5-1: new high-speed USB device number 2 using xhci_hcd
[Mo Nov 23 14:32:26 2020] usb 5-1: New USB device found, idVendor=8086, idProduct=0b64, bcdDevice=10.41
[Mo Nov 23 14:32:26 2020] usb 5-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[Mo Nov 23 14:32:26 2020] usb 5-1: Product: Intel(R) RealSense(TM) 515
[Mo Nov 23 14:32:26 2020] usb 5-1: Manufacturer: Intel(R) RealSense(TM) Camera
[Mo Nov 23 14:32:26 2020] usb 5-1: SerialNumber: 00000000F0265069
```
I've already tried the following:
- connected to two different PCs, each to two or three different USB 3 ports
- checked the ports: they are fine, because a USB 3 stick works at full speed on these ports (faster than possible with USB 2)
- also used a different USB 3 cable
- tried it on Linux and Windows (Win10: the camera disappears from the device manager after a while)

Is there any way I can fix the error? Can someone help me please?

ofl1, 2020-11-23:
The same error occurs on Linux Mint, based on Ubuntu 18.04, with Kernel 5.4.0. 

Is there any way to update the firmware to v1.5.1.3, because it supports USB2?

Any help would be greatly appreciated.

ofl1, 2020-11-29:
Error -71 always occurs when the kernel communicates with the USB3 device. -71 means protocol error. I examined it more closely with Wireshark. Here is a screenshot of it.

[image]

The host requests configurations. The device responds with some valid configuration descriptors. Some valid interface descriptors follow and finally a malformed packet. The URB announced a data block with 512 bytes. The last three bytes of this data block are the beginning of an incomplete video streaming interface descriptor. The host makes two more attempts, resets the port, and tries three more times. Then it blocks the USB3 device completely and continues via USB2. USB2 is apparently not supported by the firmware that is on the device.

Could this be a bug in the firmware? Can you find a workaround for this? Is it possible to update the firmware bypassing the kernel?

ofl1, 2020-12-01:
I made a small change in librealsense so that the L515 is recognized even though it is not entirely okay.
```
$ rs-enumerate-devices
Device info: 
    Name                          : 	Intel RealSense L515
    Serial Number                 : 	f0265069
    Firmware Version              : 	01.04.01.00
    Recommended Firmware Version  : 	01.05.02.00
    Physical Port                 : 	2-1-2
    Debug Op Code                 : 	15
    Product Id                    : 	0B64
    Camera Locked                 : 	YES
    Usb Type Descriptor           : 	2.1
    Product Line                  : 	L500
    Asic Serial Number            : 	0003a9d9cd46
    Firmware Update Id            : 	0003a9d9cd46
```
However, the firmware update does not work.
```
$ rs-fw-update -f /d1/realsense/Signed_Image_UVC_1_5_2_0.bin

Warning! the camera is connected via USB 2 port, in case the process fails, connect the camera to a USB 3 port and try again

updating device: 
Name: Intel RealSense L515, serial number: f0265069, update serial number: 0003a9d9cd46, firmware version: 01.04.01.00, USB type: 2.1

firmware update started

RealSense error calling rs2_update_firmware_cpp(device:0x7f9f840059f0, fw_image:0x7f9f8c160010):
    Device: 0003a9d9cd46 failed to download firmware
Please verify that no other librealsense application is running
```
After the failed update attempt, the device is in the recovery state. Then the device will at least be recognized as a USB3 device.
```
[  284.533804] usb 2-1: USB disconnect, device number 2
[  284.887109] usb 4-1: new SuperSpeed Gen 1 USB device number 2 using xhci_hcd
[  284.904759] usb 4-1: New USB device found, idVendor=8086, idProduct=0b55, bcdDevice= 1.00
[  284.904762] usb 4-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  284.904764] usb 4-1: Product: Intel(R) RealSense(TM) RS500 Device
[  284.904766] usb 4-1: Manufacturer: Intel 
[  284.904768] usb 4-1: SerialNumber: 111111111111
```
Even then, the update is not possible.
```
$ rs-fw-update -f /d1/realsense/Signed_Image_UVC_1_5_2_0.bin -r

update to FW: /d1/realsense/Signed_Image_UVC_1_5_2_0.bin

recovering device: 
Name: Intel RealSense L5xx Recovery, serial number: unknown, update serial number: 0003a9d9cd46, firmware version: unknown, USB type: unknown

firmware update started

failed to recover device
```
In Wireshark you can see that the device replies to the first sent 1024-byte block of the firmware with protocol error (-71).

Is the device faulty? Or can it be made to work somehow?

RealSenseSupport, 2020-12-01:
Hi @ofl1 

Sorry to hear the device isn't able to be recognized as a USB3 device.  I know you're working and operating under Linux, just confirming you're having the same problem under Windows as well?  

If possible just download the RealSense Viewer executable directly (https://github.com/IntelRealSense/librealsense/releases/download/v2.40.0/Intel.RealSense.Viewer.exe) under Windows and see if the FW update process within the Viewer itself cannot update the FW.

If the FW update process does not work, you may need to RMA the device and request a replacement.

ofl1, 2020-12-02:
Thank you for your response. Under Windows, the behavior is unchanged (see above). The RealSense Viewer reports:
```
 02/12 09:04:31,350 DEBUG [7604] (win-helpers.cpp:87) res=pSensorManager->GetSensorsByCategory(SENSOR_CATEGORY_ALL, &pSensorCollection) returned: HResult 0x80070490: "Element nicht gefunden."
 02/12 09:04:31,350 INFO [7604] (context.cpp:381) Found 0 RealSense devices (mask 255)
```
The device manager first reports "The device is working properly", and after a few minutes "This hardware device is not currently connected to the computer. (Code 45)", although the device is still connected.

I had already requested a replacement. Now the return label has finally arrived. That's a strange error, isn't it?

anhTuan0712, 2020-12-03:
> Hi @ofl1
>
> Sorry to hear the device isn't able to be recognized as a USB3 device. I know you're working and operating under Linux, just confirming you're having the same problem under Windows as [...]

Hi RealSenseSupport, could you help me to solve this problem, I expect feedback from you, thank you! 
https://github.com/IntelRealSense/librealsense/issues/7906

anhTuan0712, 2020-12-03:
> Hi @ofl1
>
> Sorry to hear the device isn't able to be recognized as a USB3 device. I know you're working and operating under Linux, just confirming you're having the same problem under Windows as [...]

Hi @RealSenseSupport 
, could you help me to solve this problem, I expect feedback from you, thank you!
#7906

laihaotao, 2020-12-31:
any update? after multiple attempts, cannot even detect the device

RealSenseSupport, 2021-01-13:
What we have been noticing is that some systems USB3 controllers aren't able to for some reason connect the L515 camera as USB3 device.  So as we've outlined in some of our whitepapers and in certain configurations that a USB3 powered hub works.  We've been suggesting to try for these issues where USB controller compatibility issues with L515 camera to use something like this hub:  https://www.amazon.com/dp/B00DQFGH80/ref=asc_df_B00DQFGH805315309/?tag=hyprod-20&creative=394997&creativeASIN=B00DQFGH80&linkCode=df0&hvadid=167151358503&hvpos=1o1&hvnetw=g&hvrand=9308811005364068537&hvpone=&hvptwo=&hvqmt=&hvdev=c&hvdvcmdl=&hvlocint=&hvlocphy=9032905&hvtargid=pla-281221827342

dqii, 2021-01-15:
I'm having this same problem with the D415

RealSenseSupport, 2021-01-27:
@dqii 

Has moving to another system with a different USB controller or using a powered USB3 hub work for you on the D415?

dqii, 2021-01-27:
@RealSenseSupport I'm only able to get it to work if I attach the USB3 cable that came with the camera to the camera. I'm able to add a hub and a USB-A->USB-C converter in between. I tried attaching multiple different USB3 cables to the camera, but none of them worked. The cable that came with the camera is too short for me, so ideally there would be a way to use a different USB3 cable.

For what it's worth, I noticed the USB3 cable that came with the camera has a cord labelled as 3.0, but that it's recognized by the software as 3.1.

RealSenseSupport, 2021-02-26:
@dqii 

Definitely need to make sure that the cable is certified USB3 cable.

ofl1, 2021-03-05:
Today a quick update from me. All advice regarding USB cables and self-powered hubs has not worked for me. I sent the device to RealSense for a firmware update so that I can at least use it in USB2 mode.

In the meantime, I bought a small PCIe card with a USB3 port and plugged it into an old PC. At this port the L515 then worked in USB3 mode!

The problem was the USB host controller. Both PCs that I tried before have ASM1042's. Two are on the mainboard of an eight-year-old PC, another one in an ExpressCard that I bought new. Controller driver and firmware updates did not help or were not possible. The PCIe card has a different USB controller and the L515 works with it.

In order to be able to use the ASM1042, I patched the kernel. I adapted the module xhci_hcd. With this change, the operating system correctly recognizes the L515 as a USB3 device. The tools realsense-viewer, rs-enumerate-devices and rs-hello-realsense then work partially, but with many warnings and errors.

Perhaps this information will help someone who also has this controller.

ofl1, 2021-03-11:
Here's a recap of how I got the camera working.

Desktop PC: PCIe card with Renesas uPD720202, tested under Linux and Windows
Laptop: ExpressCard with Fresco Logic FL1100, and self-powered hub (atolla or amazon basics), tested under Linux

The ASM1042 is not suitable. At best, the L515 runs in USB2 mode on this controller. But the resolution is then low: only 320x240 Depth and 640x480 RGB. In USB3 mode you get a lot of errors and crashes. I don't understand why this ancient controller is still being sold. RealSense support could alert users who have this controller to the problem.

RealSenseSupport, 2021-03-26:
Thanks @ofl1 - appreciate the detailed testing on the controllers.  We will keep this information available to alert others regarding this controller.

closed 2021-04-08 by RealSenseSupport