realsenseai/librealsense #7845: single_depth_color_640x480.bag violates ROSBag file format specification

MarekKnapek, 2020-11-23:
Hi, I'm in process of writing a ROSBag file format parser and I found [single_depth_color_640x480.bag](https://github.com/IntelRealSense/librealsense/blob/master/unit-tests/resources/single_depth_color_640x480.bag) file in your repository, I'm using it as a unit test for my parser.

I would like to inform you, that this file violates ROSBag file format specification.

If you look at [Bags/Format/2.0 - ROS Wiki](http://wiki.ros.org/Bags/Format/2.0) page, paragraph [Connection](http://wiki.ros.org/Bags/Format/2.0#Connection), you will see, that connection record must have specific body. In particular, the connection record body must contain a connection header. Connection header consists of mandatory and optional fields. Mandatory fields are `topic`, `type`, `md5sum` and `message_definition`.

Sadly, the file in your repository is missing the `topic` field and thus it is invalid ROSBag file.

MartyG-RealSense, 2020-11-23:
Hi @MarekKnapek   The bag format used by the RealSense SDK is mostly based on the **rosbag** format but not every rosbag may be able to be played back in the SDK.  Some have also found that an SDK-recorded bag will not play in the ROS bag playback. 
 Dorodnic the RealSense SDK Manager provides advice about this in the links below.

**Rosbag playback in SDK**
https://github.com/IntelRealSense/librealsense/issues/3020#issuecomment-452321766

**SDK bag playback in ROS**
https://github.com/IntelRealSense/librealsense/issues/4260

MartyG-RealSense, 2020-12-01:
Hi @MarekKnapek   Do you require further assistance with this case, please?   Thanks!

MarekKnapek, 2020-12-02:
Hi @MartyG-RealSense, I don't need any assistance with this.

You should be sad, that file in your repository is not readable by standard ROSBag tools, but apparently, you are not.

Feel free to close this issue with something like won't-fix / as-designed / not-a-bug resolution tag.

Marek

MartyG-RealSense, 2020-12-02:
I am sorry that you are disappointed in the design decision to focus on camera-only data.  https://github.com/IntelRealSense/librealsense/issues/3020#issuecomment-452321766

closed 2020-12-02 by MartyG-RealSense