realsenseai/librealsense #7846: What is DoF(Depth of Field) of D435 camera?

JANGSOONMYUN, 2020-11-24:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 

I would like to know what if Depth of Field (DoF) of the color camera of D435. (NOT Field of View and Degree of Freedom)
I couldn't find any information of range of camera focus in the document.

Would you let me know it?

MartyG-RealSense, 2020-11-24:
Hi @JANGSOONMYUN   I could not find any data for the Depth of Field of the D435 (or any RealSense camera model) either.  Depth of Field is something that can be calculated for a camera using a formula though:

https://en.wikipedia.org/wiki/Depth_of_field#Factors_affecting_depth_of_field

If you have access to an Android device, there is a **DOF Calculator** app on the Google Play Store.

https://play.google.com/store/apps/details?id=com.aimenrg.dof

dorodnic, 2020-11-24:
It seems to be related to camera focus, as well as camera intrinsic calibration. 
@agrunnet - perhaps you know?
@JANGSOONMYUN - in what context are you using this number?

agrunnet, 2020-11-24:
@JANGSOONMYUN  I think you want to know what the focal length of the lens is and the F# so you can calculate all this. 
The D415 and D435 share the same RGB sensor which is the OV2740 (1920x1080). The lens is a custom lens with FOV of ~70x43deg. The focal length is 1.92mm and the F# is 2.0. Assuming an allowable circle of confusion of two pixels (which are 1.4um each), you get a hyper focal distance of 66cm.  I think this is all the information you need. In general we are in focus from half the hyper focal distance (~33cm) to infinity.  The focus is fixed.

JANGSOONMYUN, 2020-11-25:
@dorodnic Thank you. I need it to make sure working distance of the camera.
@agrunnet Yes that is what I need. I couldn't find information of CoC and the hyper focal distance. Thank you.

closed 2020-11-25 by JANGSOONMYUN

MartyG-RealSense, 2020-11-25:
Thanks very much for the update @JANGSOONMYUN  - I'm pleased that we could be of help to you.