realsenseai/librealsense #7847: sdk 2.40.0 does not detect a realsense device in ubuntu 20.04

wsj6928, 2020-11-24:
| | |
|---|---|
|**librealsense**|2.40.0 RELEASE|
|**OS**|Linux|

Please provide a description of the problem

OS: ubuntu 20.04
SDK: v2.40.0
realsense device: d435
PC: intel NUC

**Problem 1. realsense-viewer can't find the device** 
/////////////////////////////////////////////////////////////////////////////
11:39:40 [Info]  - librealsense version: 2.40.0
11:39:40 [Info] /home/jenkins/workspace/LRS_SDK_CI_Debian/src/src/rs.cpp:2565 - Framebuffer size changed to 1344 x 840
11:39:40 [Info] /home/jenkins/workspace/LRS_SDK_CI_Debian/src/src/rs.cpp:2565 - Scale Factor is now 1
11:39:40 [Info] /home/jenkins/workspace/LRS_SDK_CI_Debian/src/src/linux/backend-v4l2.cpp:498 - Cannot access /sys/class/video4linux
11:39:40 [Info] /home/jenkins/workspace/LRS_SDK_CI_Debian/src/src/context.cpp:381 - Found 0 RealSense devices (mask 254)
11:39:41 [Info] /home/jenkins/workspace/LRS_SDK_CI_Debian/src/src/linux/backend-v4l2.cpp:498 - Cannot access /sys/class/video4linux
11:39:41 [Info] /home/jenkins/workspace/LRS_SDK_CI_Debian/src/src/context.cpp:381 - Found 0 RealSense devices (mask 254)
/////////////////////////////////////////////////////////////////////////////

**Problem 2. [roslaunch realsense2_camera rs_camera.launch] can't find the device**
///////////////////////////////////////////////////////////////////////////////
process[master]: started with pid [44309]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 1818a216-2dfc-11eb-a7a2-f97f5c6fbdcd
process[rosout-1]: started with pid [44319]
started core service [/rosout]
process[camera/realsense2_camera_manager-2]: started with pid [44326]
process[camera/realsense2_camera-3]: started with pid [44327]
[ INFO] [1606184633.139574394]: Initializing nodelet with 8 worker threads.
[ INFO] [1606184633.257012132]: RealSense ROS v2.2.20
[ INFO] [1606184633.257044617]: Built with LibRealSense v2.40.0
[ INFO] [1606184633.257057098]: Running with LibRealSense v2.40.0
[ WARN] [1606184633.262779723]: No RealSense devices were found!
[ WARN] [1606184639.270302165]: No RealSense devices were found!
[ WARN] [1606184645.277677589]: No RealSense devices were found!
[ WARN] [1606184651.284833220]: No RealSense devices were found!
[ WARN] [1606184657.292246758]: No RealSense devices were found!
/////////////////////////////////////////////////////////////////////////////

lsusb output:
/////////////////////////////////////////////////////////////////////////
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 013: ID 8086:0b07 Intel Corp. Intel(R) RealSense(TM) Depth Camera 435 
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 8087:0026 Intel Corp. 
Bus 001 Device 002: ID 062a:4101 MosArt Semiconductor Corp. Wireless Keyboard/Mouse
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
///////////////////////////////////////////////////////////////////////////

I tried
1. reinstalling the OS(ubuntu 20.04)
2. changing the cable
3. changing the device(d435)

MartyG-RealSense, 2020-11-24:
Hi @wsj6928   Apologies for the delay in responding, as I was considering your case very carefully.

If you have done a wipe and reinstall of Ubuntu 20 and have only installed librealsense once snce then, the problem is unlikely to be related to a conflict between multiple-installed **udev** device handling rules.

As the issue seems to be a kernel-related issue around **video4linux**, as a first step in the investigation I would recommed trying to install librealsense using the **RSUSB** method.  This method of installation requires an internet connection but bypasses video4linux, and it is not reliant on Linux versions or kernel versions and does not need patching.

The SDK release notes for 2.40.0 describe how an RSUSB installation may improve performance.

https://github.com/IntelRealSense/librealsense/wiki/Release-Notes#release-2400

wsj6928, 2020-11-25:
I changed the OS to 18.04 and now it works well

closed 2020-11-25 by wsj6928