realsenseai/librealsense #7848: Realsense2-net for C#

astroas, 2020-11-24:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 |
| Firmware Version                   | 01.05.02.00 | 
| Operating System & Version |   Windows 10  | 
| Platform                                 | PC/Raspberry Pi |
| SDK Version                            |  2.39                     | 
| Language                            |  C#                          | 
| Segment			|  Network                       | 

### realsense2-net for C# 
I am using the L515 Lidar camera to get the depth frames over the network from a Raspberry Pi. I have the L515 successfully connected to a raspberry pi working with over the network using the Intel RealSense Viewer on Windows 10 with the source connected as a network device. 
I would like to add the network device to my C# project but I cant find any examples. I have been using the [open-source-ethernet-networking-for-intel-realsense-depth-cameras](https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras) as a guide. Is this possible in C#?
Any help would be appreciated.

dorodnic, 2020-11-24:
Right now we don't include net extension in our C# wrapper, but it should be relatively easy to add if you are building from source.
You need to wrap `rs2_device* rs2_create_net_device(int api_version, const char* address, rs2_error** error);` inside https://github.com/IntelRealSense/librealsense/blob/master/wrappers/csharp/Intel.RealSense/NativeMethods.cs , similar to other methods. Note that rs2_create_net_device is exported from realsense2-net.dll and not realsense2.dll
Once native function is wrapped, you can add constructor in https://github.com/IntelRealSense/librealsense/blob/master/wrappers/csharp/Intel.RealSense/Devices/Device.cs and use it as regular device.

astroas, 2020-11-24:
Hi @dorodnic ,
I am new to creating a C# wrapper and need a little help.
I have been able to build without changes from the latest source but had to download an older package V2.12 to get the Intel.RealSense.SDK.sln and add to the latest. Intel.RealSense.SDK.sln is missing in the latest package

I have added to NativeMethod.cs in the rs_device region

`[DllImport("realsense2-net", CallingConvention = CallingConvention.Cdecl)]`

`internal static extern IntPtr rs2_create_net_device(int api_version, string address, [MarshalAs(UnmanagedType.CustomMarshaler, MarshalTypeRef = typeof(ErrorMarshaler))] out object error);`

but where do I find realsense2-net.dll?

astroas, 2020-11-25:
Hi @dorodnic,
I have been able to build realsense2-net.dll by using the instructions from [https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_windows.md](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_windows.md) and turning on the flag BUILD_NETWORK_DEVICE. 
I didn't realise I had to compile and build librealsense2.
Next step is to get rs2_create_net_device working.

AlexisC34, 2021-01-07:
Hello @astroas,

I'm currently trying to do the same thing as you and i'm new in creating wrappers.
I'm stuck in creating realsense2-net.dll, even by following the link you provided.

Can you share in details what you did to make it work please?

Thanks in advance,

Regards,
Alexis

astroas, 2021-01-08:
> Hello @astroas,
> 
> I'm currently trying to do the same thing as you and i'm new in creating wrappers.
> I'm stuck in creating realsense2-net.dll, even by following the link you provided.
> 
> Can you share in details what you did to make it work please?
> 
> Thanks in advance,
> 
> Regards,
> Alexis

Hi @AlexisC34,

Did you enable the flag BUILD_NETWORK_DEVICE ?

Regards
Andrew

AlexisC34, 2021-01-08:
Thanks for the reply!

I might be doing something wrong since i'm really new in this kind of stuff...
Here is what I do in CMake :

[image]

And in the CMakeOutput.log i just have this line : "_The system is: Windows - 10.0.18363 - AMD64_"

Regards,
Alexis

AlexisC34, 2021-01-20:
Hello @astroas,

I've been able to generate files with CMake, and then build the librealsense2.
In the \Debug folder i can find the file "realsense2-netd.dll" but not the "realsense2-net.dll" you mentioned.

Am I doing something wrong ?

astroas, 2021-01-22:
Hi AlexisC34,

Did you set the Solution Configuration to Release?

astroas, 2021-01-22:
Hi @AlexisC34,

At the top of NativeMethod.cs I have 
```
#if DEBUG
        private const string dllName = "realsense2d";
        private const string dllNameNet = "realsense2-netd";
#else
        private const string dllName = "realsense2";
        private const string dllNameNet = "realsense2-net";
#endif
```
And make sure you have the Solution Configurations set to Release.

Regards
Andrew

RealSenseSupport, 2021-02-10:
Is everything working now?  Is there anything else needed on this thread?

If we don’t hear from you in 7 days, this issue will be closed.

closed 2021-02-19 by RealSenseSupport