realsenseai/librealsense #7849: How to work around updating `udevadm` when installing librealsense on a Docker container?

lieuzhenghong, 2020-11-24:
**TL;DR: The `udevadm control --reload-rules` command won't work on a Docker container. How can I get around this?
See [this SO post](https://stackoverflow.com/questions/64752598/udevadm-in-docker-the-command-bin-sh-c-udevadm-control-reload-rules-retur) for details.**

---

I'm trying to build a Dockerfile to make building on Jetson MX simple and fuss-free (see #7722) but am having trouble doing so. The main problem is that the whole set of `udevadm` commands won't work on the Jetson. Does anyone have a workaround?

Dockerfile reproduced here:

```dockerfile
FROM python:3.8-slim

RUN apt-get update && apt-get install -y --no-install-recommends \
	python3-pip \
	python3-dev

# Install LibRealSense from source
RUN git clone https://github.com/IntelRealSense/librealsense.git
## Install the core packages required to build librealsense libs
RUN apt-get install -y git libssl-dev libusb-1.0-0-dev pkg-config libgtk-3-dev
### Distribution-specific packages for Ubuntu 18
RUN apt-get install -y libglfw3-dev libgl1-mesa-dev libglu1-mesa-dev
### Run Intel Realsense permissions script
RUN pwd
WORKDIR ./librealsense
RUN pwd
RUN ls ./config
# Make sure that your RealSense cameras are disconnected at this point
# RUN ./scripts/setup_udev_rules.sh
RUN cp ./config/99-realsense-libusb.rules /etc/udev/rules.d/
RUN apt-get install -y udev
RUN udevadm control --reload-rules
RUN udevadm trigger
# Now starting the build
RUN mkdir build && cd build
## CMake with Python bindings
## see link: https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python#building-from-source
RUN cmake ../ -DBUILD_PYTHON_BINDINGS:bool=true
## Recompile and install librealsense binaries
RUN make uninstall && make clean && make -j6 && make install
RUN export PYTHONPATH=$PYTHONPATH:/usr/local/lib/python3.6/pyrealsense2
```

MartyG-RealSense, 2020-11-24:
Hi @lieuzhenghong    I am not sure if the link below will be exactly what you need, but it does describe problems with udevadm when using Docker and a RealSense user's method for installing librealsense in a Docker container.

https://github.com/IntelRealSense/realsense-ros/issues/388#issuecomment-410672421

In the same discussion, another RealSense user took the approach of creating a **snap** package instead a a mean of simplying installation.

https://github.com/IntelRealSense/realsense-ros/issues/388#issuecomment-410786880

MartyG-RealSense, 2020-12-01:
Hi @lieuzhenghong   Do you require further assistance with this case, please?   Thanks!

lieuzhenghong, 2020-12-03:
Yes @MartyG-RealSense , I looked at the links and I still don't quite get it. Creating a snap package is out of the question for me, I'm afraid.

MartyG-RealSense, 2020-12-03:
Hi @lieuzhenghong   I researched the case further but did not find any further leads in regard to udevadm and Docker when used with librealsense.   So I tried expanding the search to be more general and look for **Nvidia** problems involving udevadm and Docker.

An example result from this expanded search was the link below:

https://github.com/NVIDIA/nvidia-docker/issues/726#issuecomment-577807936

closed 2020-12-17 by MartyG-RealSense