realsenseai/librealsense #7852: Images seem to be mosaic

Tom127, 2020-11-25:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Firmware Version                   | (v2.29.0) | 
| Operating System & Version |   { Linux (Ubuntu 16)   | 
| Kernel Version (Linux Only)    |                                      | 
| Platform                                 | PC  |
| SDK Version                            |                           | 
| Language                            |  {python }                          | 
| Segment			|  {others }                          | 

### Issue Description
Hi, @dorodnic , @MartyG-RealSense 
The images I captured with Python scripts are blurry and seem to be mosaic. Here is my script:
[test.txt](https://github.com/IntelRealSense/librealsense/files/5595327/test.txt)

Here are the images I captured with my script:
[image]
[image]
[image]

And here is the image I captured with realsense-viewer, it seems normal:

[image]

---------------------------------------------------------------------------------

Another problem is that I changed the camera settings in realsense-viewer, such as resolution, but the next time I restart realsense-viewer, it was reset.

--------------------------------------------------------------------------------

What causes the above two problems？

MartyG-RealSense, 2020-11-25:
Hi @Tom127   Are you applying a **Decimation** post-processing filter to the RGB image in Python please?   It resembles the kind of pixellation produced if the Decimation filter is set to a high value ('8' in the image below).

[image]

In regard to losing settings when restarting the Viewer, some of the settings are not persistent from one session to another.  You can store settings for a subsequent session by using the **json** saving option near the top of the Viewer's options side-panel and then loading the file back in when you next open the Viewer by using the preset load option that is nexr to the save option.

[image]

Tom127, 2020-11-26:
> Hi @Tom127 Are you applying a **Decimation** post-processing filter to the RGB image in Python please? It resembles the kind of jagged pixellation produced if the Decimation filter is set to a high value ('8' in the image below).
> 
> [image]
> 
> In regard to losing settings when restarting the Viewer, some of the settings are not persistent from one session to another. You can store settings for a subsequent session by using the **json** saving option near the top of the Viewer's options side-panel and then loading the file back in when you next open the Viewer by using the preset load option that is next to the save option.
> 
> [image]

Hi @MartyG-RealSense ，thanks for your reply. but, whether I use post-processing filters in the python script or not, the images still look like a mosaic. I closed the filter in realsense-viewer, and then ran this python script, the result was still the same. This is all the code I tested. Could you run it and see the images. 

[test.txt](https://github.com/IntelRealSense/librealsense/files/5601918/test.txt)

MartyG-RealSense, 2020-11-26:
I had a look at your kindly provided code.  If align is applied before post-processing then a distortion effect called **aliasing** may occur that causes jagged lines.  So it is recommended that align is applied **after** post-procesing.

https://github.com/IntelRealSense/librealsense/issues/1207#issuecomment-367718085

So it may be worth re-arranging your code if align is currently applied before the filters as a matter of best-practice, though I note your mention that removing post-processing filters does not affect whether the problem occurs.

Tom127, 2020-11-27:
> I had a look at your kindly provided code. If align is applied before post-processing then a distortion effect called **aliasing** may occur that causes jagged lines. So it is recommended that align is applied **after** post-procesing.
> 
> [#1207 (comment)](https://github.com/IntelRealSense/librealsense/issues/1207#issuecomment-367718085)
> 
> So it may be worth re-arranging your code if align is currently applied before the filters as a matter of best-practice, though I note your mention that removing post-processing filters does not affect whether the problem occurs.

I found that if I change the img_size in the Python script to [640,480], the image is normal, but if I change the img_size to [848,480], the image appears jagged, same code as yesterday, no post-procesing filter. What is the problem and how to solve it?
By the way, I have no idea how to apply  align after post-processing, could you provide complete example code? Thanks.

MartyG-RealSense, 2020-11-27:
I would recommend changing the depth scale from 0.0001 in your Python script to the camera's default of **0.001** (which is the default that the RealSense Viewer also uses when launched).   

**depth_sensor.set_option(rs.option.depth_units, 0.001)**

I would also add that if you are performing depth to color alignment in your Python script, comparing an aligned image from your Python script with an RGB image from the Viewer would not be a like-for-like comparison, because the Viewer's **2D** mode does not have the ability to align depth and color (the **3D** point cloud mode of the Viewer does though).  So the Viewer in 2D mode would produce a pure RGB image like a video webcam.

The Python script in the link below has an example of setting up post-processing first and then performing **align_to** afterward.

https://github.com/IntelRealSense/librealsense/issues/2870#issuecomment-446195769

Tom127, 2020-12-03:
> I would recommend changing the depth scale from 0.0001 in your Python script to the camera's default of **0.001** (which is the default that the RealSense Viewer also uses when launched).
> 
> **depth_sensor.set_option(rs.option.depth_units, 0.001)**
> 
> I would also add that if you are performing depth to color alignment in your Python script, comparing an aligned image from your Python script with an RGB image from the Viewer would not be a like-for-like comparison, because the Viewer's **2D** mode does not have the ability to align depth and color (the **3D** point cloud mode of the Viewer does though). So the Viewer in 2D mode would produce a pure RGB image like a video webcam.
> 
> The Python script in the link below has an example of setting up post-processing first and then performing **align_to** afterward.
> 
> [#2870 (comment)](https://github.com/IntelRealSense/librealsense/issues/2870#issuecomment-446195769)

Hi, @MartyG-RealSense @ev-mp @dorodnic even though I modified the depth scale, the image didn't get clear. Then I found that the problem was in line 40-42 of my python script. I deleted lines 41 and 42 (`pipeline.stop(), profile = pipeline.start(config)`), and the image became clear again, but every time I opened the terminal and ran the script for the first time, the XYZ value read from the camera was totally wrong, It is necessary to close the program at the terminal and rerun the program, then the XYZ-data will be normal. (which is why I stop and restart pipeline in the script, lines 41 and 42). What's the problem? Is there something wrong with the SDK? Here is my Python script.
[test.txt](https://github.com/IntelRealSense/librealsense/files/5634237/test.txt)

MartyG-RealSense, 2020-12-03:
Removing those two lines is what I was going to suggest, so I believe that you are on the right track with that approach.  The question is how to rearrange the stop mechanism.

What happens if you put the jpg writing code and pipeline.stop() inside the **If** check for the keypress so that the image is saved to jpg and then the pipeline is stopped if the 'q' key is pressed to exit the process?

**if(cv2.waitKey(1) & 0xFF) == ord('q'):
print("Exit...")
cv2.imwrite("z.jpg", color_image)
pipeline.stop()**

closed 2020-12-17 by MartyG-RealSense