realsenseai/librealsense #7855: D415 Depth Origin Location Consistancy

dexter-chen, 2020-11-25:
### Issue Description
<
Anyone met similar issue as me?

According to realsense datasheet, the depth origin of D415 shall be the center of left imager, but during our test not all the device meets this SPEC. Some of their origin locates in the center of right imager, some locates in the center of RGB imager, some them are even out side the camera.

Our test method is to mount the D415 vertically on a wall with guiding of a laser level-meter. The right imager is on the top, then the left imager and then the RGB imager is at the bottom. Then we project a horizontal laser line with this level-meter and make the laser line to be at the same height of left imager of D415. Then we read the laser line pixel location. It shall be same as the principle X,Y. But actually, some times they are, some times they are not.

Anyone knows how to make the origin of D415 depth locate in the center of left imager as the datasheet described? Any calibration method?

Thank you.
>

MartyG-RealSense, 2020-11-25:
Hi @dexter-chen   The origin should consistently be at the center of the left imager, yes.

It sounds as though you are rotating the camera on its side so that the RGB imager is at the bottom.  Can you confirm this, please?

Given your description of the vertical arragement of the sensors from bottom to top, it seems that you are also aware of the left imager being on the right side of the D415 camera when observing the camera from a frontal viewpoint, with the RGB sensor on the furthest right.

https://github.com/IntelRealSense/librealsense/issues/3662

As a starting point in this case, may I ask whether your system expects that when the camera is rotated on its side, the camera image rotates as well.  On the 400 Series cameras, image rotation does not take place when rotating the camera on its side.

https://github.com/IntelRealSense/librealsense/issues/6023#issuecomment-597615585

dexter-chen, 2020-11-26:
@MartyGt, Thanks for your quick Reply.

Yes. We rotate the camera as you described.

For our system, we will rotate the camera (90 degree CW when observing the camera from frontal view point) and mount it on the wall, while we install the system. After that we will keep the camera orientation. We also rotate both RGB and Depth images to keep up/down direction consistent with the physical world.  For our case, we directly apply default principle X value as our new principle Y value. We also apply Intel SDK to make the RGB and Depth Image alignment. Then we read the laser line position from the aligned RGB image in image coordinates and assume it shall also be the same position of depth image.

Is there anything incorrect and cause this issue?

Thank you.

MartyG-RealSense, 2020-11-26:
I'm not certain that rotating the image will correctly update the SDK's XY coordinate system due to how it works, using a system of plus and minus horizontal and vertical values with zero as the center of the left imager.  The subject is discussed in detail in the link below.

https://github.com/IntelRealSense/librealsense/issues/7279

MartyG-RealSense, 2020-12-03:
Hi @dexter-chen   Do you require further assistance with this case, please?   Thanks!

dexter-chen, 2020-12-08:
@MartyG-RealSense Yes. I'm testing more samples. And I may go back to you later next week.

MartyG-RealSense, 2020-12-08:
Okay, thanks very much @dexter-chen for the update - I look forward to hearing from you later!

MartyG-RealSense, 2020-12-15:
Adding a note to keep this case open for a further time period.

MartyG-RealSense, 2020-12-28:
Hi @dexter-chen   Do you have an update that you can provide, please?   Thanks!

MartyG-RealSense, 2021-01-03:
Case closed due to no further comments received.

closed 2021-01-03 by MartyG-RealSense

koalaaaaaaaaa, 2022-03-26:
@MartyG-RealSense 
Hi Marty, what about D455 camera, the origin of that is still left   @@imager?
[image]

MartyG-RealSense, 2022-03-26:
Hi @koalaaaaaaaaa   Yes, the center-line of the left imager is still the depth origin on D455.