realsenseai/librealsense #7856: What do gamma correction values mean for the RGB value?

RealitiesMediaCube, 2020-11-25:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 

### Issue Description

For some real sense cameras, developers can set the gamma correction value of the realsense cameras, e.g., for the D435 between 100 and 500 according to the data sheet. 

I need to extract linear RGB data without gamma correction though, ideally by deactivating gamma correction or removing gamma correction from the RGB values of the camera.

My question is, what does the gamma value range mean? Does the range 100 to 500 represent floating point values? E.g., a gamma correction of 100 means 1.00, i.e., no gamma correction? Then I could just use this setting for uncorrected RGB values.

Thanks in advance.

MartyG-RealSense, 2020-12-03:
Hi @RealitiesMediaCube   Do you require further assistance with this case, please?   Thanks!

RealitiesMediaCube, 2020-12-03:
Hi @MartyG-RealSense,
thanks for the info about the raw bayered RGB images. I will look into that. 

I would appreciate any input your team can give me on the configuration of the gamma value and its impact on the image. The realsense documentation is not clear about how gamma is applied and what the range of values means.

MartyG-RealSense, 2020-12-03:
Gamma can be configured with the **RS2_OPTION_GAMMA** instruction.  A script in the link below provides an example of the use of the instruction to configure a gamma value between 300 and 500:

https://github.com/IntelRealSense/librealsense/issues/6560#issuecomment-641869430

[image]

I could not find documentation that provided further descriptive explanation.  In general though, reducing RGB gamma darkens the RGB image and increasing gamma brightens the image.  Below is shown a scene with 100 gamma (minimum) in the upper image and 500 gamma (maximum) in the lower image.

RGB gamma is independent of the RGB auto-exposure.  So whilst changing exposure and gain whikst auto-exposure is active disables the auto-exposure, changing the gamma does not affect the status of auto-exposure.

[image]

tatze82, 2020-12-03:
Thanks for the detailed information and your support @MartyG-RealSense. 

I need a linear color response from the camera for further processing. Hence, I either turn gamma of, or know how it is applied, so that I can reverse the process. 
I determined from the data sheet that the camera uses the OV9282 image sensor. Theoretically, the gamma correction can even be turned off for some image sensors, maybe also for this one. Is there a possibility in the realsense SDK to send instructions directly to the image sensor, some kind of raw input signals for controlling the sensor? 

I could also contact the chip designers directly for details on gamma correction.

RealitiesMediaCube, 2020-12-03:
Thanks for the detailed information and your support @MartyG-RealSense.

I need a linear color response from the camera for further processing. Hence, I either turn gamma of, or know how it is applied, so that I can reverse the process.
I determined from the data sheet that the camera uses the OV9282 image sensor. Theoretically, the gamma correction can even be turned off for some image sensors, maybe also for this one. Is there a possibility in the realsense SDK to send instructions directly to the image sensor, some kind of raw input signals for controlling the sensor?

I could also contact the chip designers directly for details on gamma correction.

MartyG-RealSense, 2020-12-03:
The camera sensors can be accessed directly by using callbacks with the **Low Level API**.

https://github.com/IntelRealSense/librealsense/blob/master/doc/api_arch.md#low-level-device-api

**Low-level “sensor” callbacks API calls:**
-  Minimum SDK-imposed overhead == minimum latency.
-  Each frame is handled with a separate callback - “first come – first serve”.
-  All the synchronization stuff is left for the user side to implement.
-  The **rs-data-collect** tool of the SDK uses low-level sensor API to minimize software-imposed latencies.   It is useful for performance profiling

https://github.com/IntelRealSense/librealsense/tree/master/tools/data-collect

RealitiesMediaCube, 2020-12-10:
@MartyG-RealSense I couldn't solve the problem yet. I will have to check for alternative camera solutions. I think you can close the issue, if there is no more information. Thank you for you support.

MartyG-RealSense, 2020-12-10:
I just did some further research of your problem.  I found a comment where the Chief Technical Officer of the RealSense Group at Intel (agrunnet) states that if a preset file is loaded that does not provide settings for particular options then the camera uses the previous settings.  

This would suggest that you may be able to achieve your goal of no gamma if you define a preset file that sets '0' gamma and then load the file in to apply the defined settings when your program launches.

https://github.com/IntelRealSense/librealsense/issues/1815#issuecomment-393936631

It looks as though gamma is set in the json with **controls_color_gamma**

https://github.com/IntelRealSense/librealsense/issues/3277#issuecomment-465066625

In the above link, the control_color settings for Hue and Brightness are set to zero, suggesting that gamma should be able to be zero too (I do not have confirmation of this though).

RealitiesMediaCube, 2020-12-17:
Hi, that could be an option, thanks for checking it out. I will have to try that out. 

However, without knowing how gamma is actually influenced by the value, it likely does not solve my use case. I really need to know for sure that it is on or off, or which relationship exists so that I can remove the gamma correction from the image.

MartyG-RealSense, 2020-12-17:
I carried out further research into your case.   Ultimately, if you want the certainly of obtaining raw images before they undergo processing then I believe that the earlier mentioned method of accessing and converting raw color frames may be the best option.

https://github.com/IntelRealSense/librealsense/issues/7856#issuecomment-733776542

RealitiesMediaCube, 2020-12-18:
Thanks I will try that.

closed 2020-12-18 by RealitiesMediaCube