realsenseai/librealsense #7858: wait_for_frames cannot be called if a callback was provided

OneQuestion, 2020-11-25:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435i } | 
| Firmware Version                   | (2.40) | 
| Operating System & Version |   {Ubuntu 16 | 
| Kernel Version (Linux Only)    |                                         | 
| Platform                                 | PC |
| SDK Version                            | }                          | 
| Language                            |  {C++ }                          | 
| Segment			|                       | 

### Issue Description
I am trying to use "motion" example and add depth frames. I tried adding to this example `cfg.enable_stream(RS2_STREAM_DEPTH, 848, 480, RS2_FORMAT_Z16, 30);`, but after adding `auto frames = pipe.wait_for_frames();` I get `terminate called after throwing an instance of 'rs2::wrong_api_call_sequence_error        wait_for_frames cannot be called if a callback was provided` error. How can I access the depth frames here?

MartyG-RealSense, 2020-11-26:
Hi @OneQuestion   I hope that the information in the link below will be helpful to you.

https://github.com/IntelRealSense/librealsense/issues/2945

As the discussion in the above link mentions, the **rs-callback** example program can also provides guidance for using depth and IMU data in the same program.

https://github.com/IntelRealSense/librealsense/blob/master/examples/callback/

The program's notes state: "The default video configuration contains Depth and Color streams.  If a device is capable to stream IMU data, both Gyro and Accelerometer are enabled by default".

MartyG-RealSense, 2020-12-03:
Hi @OneQuestion   Do you require further assistance with this case, please?   Thanks!

OneQuestion, 2020-12-03:
hi. I tried it and so far I couldnt get depth frames and imu data.

MartyG-RealSense, 2020-12-03:
Are you experiencing the same error as before please?

**terminate called after throwing an instance of 'rs2::wrong_api_call_sequence_error wait_for_frames cannot be called if a callback was provided**

OneQuestion, 2020-12-05:
Yes. I am trying to modify motion example, by puting *auto frames =
pipe.wait_for_frames(); *in while loop. I just want to be able to get both
frames and IMU data. But this callback thing is not letting me do that.
Maybe you could tell me how to modify this motion example to get *frames*?
Then I could adapt it to my main code. Thank you.

MartyG-RealSense, 2020-12-05:
I researched your case further and found a past C++ case about adding depth to **rs-motion** where the RealSense user used the same cfg setup code for the depth stream that you did.  Yet it was successful for them.  The details of that case suggest that adding the cfg line was the only change that needed to be made to rs-motion.

https://github.com/IntelRealSense/librealsense/issues/4185

As you mentioned earlier, it may be the use of the **auto frames = pipe.wait_for_frames()** line that is causing the problem with your particular script.  It is not present in the original rs-motion.cpp script.  The rs-motion script uses callbacks.  In a link that RealSense team member **ev-mp** provides, he explains that callbacks are **Asynchronous** and wait_for_frames is **Synchronous**, and the two cannot be mixed together.

https://github.com/IntelRealSense/librealsense/issues/3817#issuecomment-485663496

OneQuestion, 2020-12-06:
So I have to get frames in asynchronous mode? Should I use this callback method? Isn't there a simpler way to get both depth and IMU?

MartyG-RealSense, 2020-12-06:
The impression that I get from the advice in https://github.com/IntelRealSense/librealsense/issues/4185#issuecomment-501691375 is that you should only need to add the cfg statement for the depth stream.   If you only add the cfg line for depth to rs-motion and add nothing else, does the program work for you please?

OneQuestion, 2020-12-06:
> The impression that I get from the advice in #4185 (comment)

The program works with added cfg line. But then I try to get frames and
thats where I get the error.

MartyG-RealSense, 2020-12-06:
I looked at your problem carefully.  It would be best to refer your question to @ev-mp  though, as asynchronous programming is outside of my knowledge.

@ev-mp   Could you assist @OneQuestion regarding how to get depth frames on a modified **rs-motion** program with depth added via a cfg stream definition please?   Thanks!

MartyG-RealSense, 2020-12-20:
Case closed due to no new comments received.

closed 2020-12-20 by MartyG-RealSense