realsenseai/librealsense #7860: Device Enumeration crashes(nullptr access) after device stream open (Windows w/ RSUSB ON)

MojamojaK, 2020-11-26:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Operating System & Version |  Win10 | 
| Platform                                 | PC |
| SDK Version                            |  3.40.0 | 
| Language                            |  C++  | 

### Issue Description

Device enumeration using rs2::context.query_devices() to already-opened cameras crash due to nullptr access.
Does not occur with RSUSB OFF.

Root cause:
https://github.com/IntelRealSense/librealsense/blob/34fc284d537a4b873a37b737a61f1f1da92dbb60/src/winusb/handle-winusb.h#L36-L57

returns `sts != RS2_USB_STATUS_SUCCESS`

https://github.com/IntelRealSense/librealsense/blob/34fc284d537a4b873a37b737a61f1f1da92dbb60/src/winusb/device-winusb.cpp#L87-L88

throws `winapi_error("WinUsb action failed, " + GetLastError()); ` and caught here:

https://github.com/IntelRealSense/librealsense/blob/34fc284d537a4b873a37b737a61f1f1da92dbb60/src/winusb/enumerator-winusb.cpp#L165-L169

returns `nullptr`

https://github.com/IntelRealSense/librealsense/blob/34fc284d537a4b873a37b737a61f1f1da92dbb60/src/ds5/ds5-device.cpp#L566-L568

`backend.create_uvc_device(info)` becomes a smart pointer pointing to nullptr

https://github.com/IntelRealSense/librealsense/blob/34fc284d537a4b873a37b737a61f1f1da92dbb60/src/ds5/ds5-device.cpp#L573

https://github.com/IntelRealSense/librealsense/blob/34fc284d537a4b873a37b737a61f1f1da92dbb60/src/backend.h#L694-L697
`elem` is nullptr and crashes

I think librealsense should at least throw a proper exception or something so code wouldn't crash

I have provided a sample:
Program crashes on comment location.

```C++
#include <cstdlib>
#include <vector>
#include <iostream>

#include <librealsense2/rs.hpp>

typedef struct stream_profile_config
{
	int fps;
	rs2_stream stream_type;
	int stream_index;
	rs2_format format;
	int width;
	int height;

} stream_profile_config;

std::tuple<bool, rs2::stream_profile> get_stream_profile(const rs2::sensor& sensor, const stream_profile_config& config)
{
	const std::vector<rs2::stream_profile> stream_profiles = sensor.get_stream_profiles();
	for (const rs2::stream_profile& stream_profile : stream_profiles)
	{
		if (stream_profile.is<rs2::video_stream_profile>())
		{
			const rs2::video_stream_profile video_stream_profile = stream_profile.as<rs2::video_stream_profile>();
			if (video_stream_profile.fps() == config.fps
				&& video_stream_profile.stream_type() == config.stream_type
				&& video_stream_profile.stream_index() == config.stream_index
				&& video_stream_profile.format() == config.format
				&& video_stream_profile.width() == config.width
				&& video_stream_profile.height() == config.height)
			{
				return std::make_tuple(true, stream_profile);
			}
		}
	};
	return std::make_tuple(false, stream_profiles[0]);
}

int main(void)
{
	std::cout << "[INFO] Starting Realsense Sandbox Program. " << std::endl;
	std::cout << "[INFO] Librealsense Library Version: " << RS2_API_VERSION_STR << std::endl;

	{
		rs2::context context;

		const rs2::device_list device_list = context.query_devices();
		if (device_list.size() < 1)
		{
			std::cout << "[ERROR] NO REALSENSE DEVICES FOUND!" << std::endl;
			return EXIT_FAILURE;
		}

		const rs2::device device = device_list[0];

		const rs2::depth_sensor depth_sensor = device.first<rs2::depth_sensor>();

		const std::tuple<bool, rs2::stream_profile> depth_stream_profile_result = get_stream_profile(depth_sensor, { 6, rs2_stream::RS2_STREAM_DEPTH, 0, rs2_format::RS2_FORMAT_Z16, 1280, 720 });

		const rs2::stream_profile& depth_stream_profile = std::get<1>(depth_stream_profile_result);

		depth_sensor.open(depth_stream_profile);

		{
			const rs2::device_list inner_devices = context.query_devices();
			const rs2::device inner_device = inner_devices[0]; // !!!CRASHES HERE!!!
			const std::string serial_number = inner_device.get_info(rs2_camera_info::RS2_CAMERA_INFO_SERIAL_NUMBER);
			std::cout << "[INFO] Inner Serial Number: " << serial_number << std::endl;
		}
		
		depth_sensor.close();
	}

	return EXIT_SUCCESS;
}
```

MartyG-RealSense, 2020-11-26:
Hi @MojamojaK   Are you able to catch the error if you add to your test script the C++ exception handling code in the SDK's **Error Handling Scheme** documentation, please?

https://dev.intelrealsense.com/docs/librealsense-error-handling-scheme

MojamojaK, 2020-11-27:
As I mentioned, the error is a memory access violation, so no exceptions are thrown

MartyG-RealSense, 2020-11-27:
May I confirm please the method that you are using to install the Windows version of the librealsense SDK.  Are you using the executable installer file on the SDK **Releases** page:

https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_windows.md

Or are you building the SDK in Windows from source code and setting the CMake build flag **FORCE_RSUSB_BACKEND** to true?

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_windows.md

According to the CMake buid flag documentation, **FORCE_RSUSB_BACKEND** is a single instruction that can be used instead of the three different flags for different situations that there were in the past (FORCE_LIBUVC, ANDROID_USB_HOST_UVC and FORCE_WINUSB_UVC).  It allows a WinUSB rewrite of the UVC protocol to be used instead of Windows Media Foundation.

https://github.com/IntelRealSense/librealsense/wiki/Build-Configuration

Notably, the old flag for Windows - FORCE_WINUSB_UVC - required installation of a special driver provided inside the Win7 SDK installer.  I wonder if that is what is missing in your installation that could be causing problems when RSUSB is set to true).

[image]

@dorodnic can you confirm whether or not this Win7 driver is still needed if using **FORCE_RSUSB_BACKEND** on a CMake build on Windows 10, please?

MojamojaK, 2020-11-29:
Yes, I am able to enumerate devices so FORCE_RSUSB_BACKEND is properly enabled with the proper drivers.

You should try running the sample code. It will crash.

MojamojaK, 2020-11-29:
and yes, you need to install the Win7 driver to use librealsense with RSUSB_BACKEND on.

MartyG-RealSense, 2020-11-29:
Thanks very much @MojamojaK for the additional details.  Whilst I cannot make comment on any future decisions that might be taken about RSUSB on Windows, the developers are aware of the subject.

MartyG-RealSense, 2020-12-05:
Hi @MojamojaK   Do you require further assistance with this case, please?   Thanks!

MojamojaK, 2020-12-05:
No, but this is a bug and I think should be left opened until resolved

MartyG-RealSense, 2020-12-05:
Given that I have not seen another report of this particular issue, I will close it for now and re-open it if another case with the same issue is reported.

closed 2020-12-05 by MartyG-RealSense

MojamojaK, 2020-12-05:
alright.
I'll open up a pull request for a temporary workaround.
Might try to fix it later.