realsenseai/librealsense #7861: rs2_get_optoin(options:0xad811f08, option:Exposure): get_xu(id=3) failed last error: resource temporarily unavailable

mbobinger, 2020-11-26:
Camera model: D435i
Firmware version: 05.12.06.00
Firmware Update ID: 020423022673
System: Raspberry Pi4 - 4GB
Framework: Python3.6
Error message: rs2_get_option(options:0xad811f08, option:Exposure): get_xu(id=3) failed last error: resource temporarily unavailable

Dear Realsense-fellows,
I've been using the D435i-intel cams on my RPI4 for months and I haven't had many issues besides sometimes I couldn't set the power state, in that cases a sudo reboot or unplugging and replugging of the USB cam would fix the issue. However, in almost all cases I cannot access the installations because they are remote.

Since yesterday I am facing a new problem: the error mentioned above. I've done some basic homework and searched for the issue and came across following threads:
C++ fix: https://github.com/IntelRealSense/librealsense/issues/2161#issuecomment-408779150
Python fix: https://github.com/IntelRealSense/librealsense/issues/5428
C++ USB reset: https://github.com/themancalledjakob/usbreset

So I've tread the Python fix (thread 5428) and applied some simple prints for the devices, the only plugged in intel cam was also recognized. However, after that reset I still needed a sudo reboot to be able to access the device again.

So currently I am wondering what is kind of the best practice and most robust way to handle such errors?
I have thought about following code, which I tested in principle and it works (It tries to initialize the cam for 5 times and if it fails the RPI will be rebooted):

`import pyrealsense2 as rs2
import os
intel_boolean = False
for i in range(5):
    if intel_boolean == False:
        try:
            pipeline = rs.pipeline()
            config = rs.config()
            config.enable_stream(rs.stream.depth, x, y, rs.format.z16, fps)
            intel_boolean = True
        except:
            ctx = rs2.context()
            list = ctx.query_devices()
            for dev in list:
                # serial = dev.query_sensors()[0].get_info(rs2.camera_info.serial_number)
                # compare to desired SN
                dev.hardware_reset()
                intel_boolean = False
    else:
        break
if intel_boolean == False:
    print("Checked and reset the camera " + str(i) + "times without success, now rebooting")
    os.system("sudo reboot")
else:
    print("Camera worked fine after so many iterations" + str(i))

`

MartyG-RealSense, 2020-11-26:
Hi @mbobinger  All of the reset methods that you referenced (hardware_reset, USB port reset and computer reboot) are valid.  There are some circumstances though where a hardware_reset or USB port reset do not correct a problem.  So a computer reboot like the one in your script above is a dependable method of recovery from unforseen problems that might manifest.

It may be worth testing though whether a hardware_reset of the camera only instead of a full reboot of the Pi can provide a level of reliability that is satisfactory to you.  This could be done by adding **exception handling** to your script that initiates hardware_reset if an error in general is detected (not a specific error).  

The Python wrapper tutorial script below has an example of error handling code.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/python-tutorial-1-depth.py#L38

mbobinger, 2020-11-27:
@MartyG-RealSense thank you for your reply. 
I have checked and implemented my code example above but the system is still not reliable (problems everyday).
We are currently looking for a fast solution for our customer. The D435i has been running for 3months without issues, then problems occured at the startup with setting the power state, which could be fixed with a simple reboot and lately, the camera also gets stuck occasionally during operation. We somehow need to react fast and think it could be related to the intel hardware eventually.

Upon start we run a mystartupscript.desktop file in /home/pi/.config/autostart with a simple command line:
https://learn.sparkfun.com/tutorials/how-to-run-a-raspberry-pi-program-on-startup/method-2-autostart
Minimum code example:
```
[Desktop Entry]
Type=Application
Name=Realsense
Exec=/usr/bin/python3 /home/pi/counter.py
```
For troubleshooting, I'd like to generate a logfile for the .desktop script. Is there anything else I can log in the actual counter.py file where all the processing and streaming of the intelrealsense happens?

https://stackoverflow.com/questions/34812798/python-save-output-to-file-when-run-at-startup-raspberry-pi

Thank you!

MartyG-RealSense, 2020-11-27:
Assuming that you are using ordinary Python for your **counter.py** script and not running it within the RealSense pyrealsense2 wrapper, the official Python documentation for logfile creation may be helpful:

https://docs.python.org/3/howto/logging.html#logging-to-a-file

f you have the camera plugged into a USB hub then I would recommend trying it with the camera plugged directly into the computer's USB port if you have a port available, as USB hubs may sometimes cause power-related issues for the camera during boot-up.

MartyG-RealSense, 2020-12-04:
Hi @mbobinger  Do you require further assistance with this case, please?   Thanks!

mbobinger, 2020-12-04:
@MartyG-RealSense thank you marty all good for now, the hardware reset option helps me a lot. Please close the thread if you like.

MartyG-RealSense, 2020-12-04:
That's great to hear @mbobinger - thanks for the update!   I will close the case now.

closed 2020-12-04 by MartyG-RealSense