realsenseai/librealsense #7862: Failed Installing librealsense2-dkms on Jetson Nano

Schloern93, 2020-11-26:
Hello,

When I try to install the package librealsense2-dkms I get the following error: 
Unable to locate package librealsense2-dkms
All other packages I could install successfully according to the instructions.
I have already read the problem for #7310. Is it possible to install the package for ros-dashing?

Thanks in advance

MartyG-RealSense, 2020-11-26:
Hi @Schloern93   THe DKMS package installation is for **X86 / AMD64**-based devices, whilst Jetson has an **Arm** processor.  So Jetson owners should install librealsense by a different method.

An installation method that has been successful for a number of Jetson Nano users is described in the link below.

https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-707501049

Schloern93, 2020-12-02:
Thanks that worked for me

closed 2020-12-02 by Schloern93

MartyG-RealSense, 2020-12-02:
Great news @Schloern93 - thanks very much for the update!