realsenseai/librealsense #7863: SDK installation problem

TouchDeeper, 2020-11-26:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400  | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Ubuntu 16  | 
| Kernel Version (Linux Only)    |  4.15.0                                    | 
| Platform                                 | PC |
| SDK Version                            |  2.17.1. }                          | 
| Language                            |  C                      | 
| Segment			| Robot                | 

### Issue Description
Hello,
I install the realsense 2.17.1 according to the manual installation guide.
When I run:
`./scripts/patch-realsense-ubuntu-lts.sh`, a fail information occur, the total output is  below:
```
Package required libusb-1.0-0-dev:  - found
Package required libssl-dev:  - found
Create patches workspace in ubuntu-xenial-hwe folder

The kernel has modified files:
	modified:   drivers/media/usb/uvc/Makefile
	modified:   drivers/media/usb/uvc/uvc_driver.c
	modified:   drivers/media/v4l2-core/v4l2-ioctl.c
	modified:   include/uapi/linux/videodev2.h
Proceeding will reset all local kernel changes. Press 'n' within 10 seconds to abort the operation
Do you want to proceed? [Y/n]Y
Update the folder content with the latest from mainline branch
remote: Total 0 (delta 0), reused 0 (delta 0)
From git://kernel.ubuntu.com/ubuntu/ubuntu-xenial
 * branch            hwe        -> FETCH_HEAD
Resetting local changes in ubuntu-xenial-hwe folder
 HEAD is now at 50ec9ec UBUNTU: Ubuntu-hwe-4.15.0-126.129~16.04.1
Applying realsense-uvc patch
patching file drivers/media/usb/uvc/Makefile
patching file drivers/media/usb/uvc/uvc_driver.c
Hunk #1 succeeded at 208 with fuzz 2 (offset 5 lines).
patching file drivers/media/usb/uvc/uvcvideo.h
Hunk #1 FAILED at 153.
1 out of 1 hunk FAILED -- saving rejects to file drivers/media/usb/uvc/uvcvideo.h.rej
patching file drivers/media/v4l2-core/v4l2-ioctl.c
Hunk #1 succeeded at 1248 (offset 2 lines).
patching file include/uapi/linux/videodev2.h
Hunk #1 succeeded at 680 (offset 12 lines).
```
I ignore it and keep install the librealsense. After installation, I  test the installation by realsense-viewer. some warning occur:
```
 26/11 22:43:59,800 WARNING [140147348924160] (sensor.cpp:338) Unregistered Media formats : [ UYVY ]; Supported: [ ]
 26/11 22:43:59,809 WARNING [140147348924160] (backend-v4l2.cpp:1248) Pixel format 36315752-1a66-a242-9065-d01814a likely requires patch for fourcc code RW16!
 26/11 22:43:59,810 WARNING [140147348924160] (sensor.cpp:338) Unregistered Media formats : [ RW16 ]; Supported: [ ]
 26/11 22:44:04,857 WARNING [140147348924160] (types.cpp:57) set_pu(id=9) failed! Last Error: Input/output error
 26/11 22:44:04,857 WARNING [140147348924160] (sensor.cpp:722) Exception was thrown when inspecting properties of a sensor
 26/11 22:44:04,857 ERROR [140147348924160] (types.h:214) xioctl(VIDIOC_G_CTRL) failed Last Error: Invalid argument
 26/11 22:44:04,858 WARNING [140147348924160] (sensor.cpp:722) Exception was thrown when inspecting properties of a sensor
```
I check on the `Motion module`, an error occur:`Motion Module failure`, the terminal in which realsense-viewer run output:
```
 26/11 22:45:06,480 WARNING [140147215537920] (ds5-timestamp.cpp:174) HID timestamp not found! please apply HID patch.
```
Then I check on the `RGB Camera`, the terminal in which realsense-viewer run output:
```
26/11 22:53:21,611 WARNING [140146921957120] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to installation chapter for details.
```
Then I check on the `Stereo Module`, the terminal in which realsense-viewer run output:
```
26/11 22:54:33,046 WARNING [140146913564416] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to installation chapter for details.
```
Then I check off the `Motion Module`, the terminal in which realsense-viewer run output:
```
 26/11 22:55:24,067 ERROR [140147638209152] (types.h:214) Failed to enable_sensor /sys/devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1:1.5/0003:8086:0B3A.0003/HID-SENSOR-2000e1.3.auto/enable_sensor Last Error: Permission denied
```

Could you please provide some advice to solve these issues? Thanks a lot.

MartyG-RealSense, 2020-11-26:
Hi @TouchDeeper   Given the number of errors that you have experienced, I would recommend trying a **backend** installation of librealsense, which requires an internet connection but is not depenent on kernel versions or Linux versions and does not require patching.

The librealsense version **2.17.0** from November 29 2018 that you are using is very old, as the SDK is currently up to **2.40.0** at the time of writing this.  If you need to use this version for your project then the **libuvc backend** method of backend installation should be the suitable one for that age of SDK.

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

If you are able to upgrade your project to a modern SDK, then versions **2.30.0** and newer can instead use a backend installation method called **RSUSB** to build librealsense from source code with the CMake build flag **-DFORCE_RSUSB_BACKEND=true**.  Again, an internet connection is required for this method.

https://github.com/IntelRealSense/librealsense/issues/6432#issuecomment-631484711

TouchDeeper, 2020-12-01:
@MartyG-RealSense 
Thank you for your reply. Is the realsense-ros compatible with the backend installation? Are there some differences in using between manually installation with backend installation?

MartyG-RealSense, 2020-12-01:
The ROS wrapper can work with librealsense whether it is built manually or with RSUSB.

RSUSB does have limitations compared to a manual build.  For more information about this, please visit the link below and scroll down through the linked-to comment to the section headed **What are the advantages and disadvantages of using libuvc vs patched kernel modules?** 

https://github.com/IntelRealSense/librealsense/issues/5212#issuecomment-552184604

chia-labradorsystems, 2020-12-01:
`patching file drivers/media/usb/uvc/uvcvideo.h
Hunk #1 FAILED at 153.
1 out of 1 hunk FAILED -- saving rejects to file drivers/media/usb/uvc/uvcvideo.h.rej`

We have the same issue and we find out that you need to change
scripts/realsense-camera-formats-xenial-hwe.patch
in order to match a recent change of uvcvideo.h
 
[realsense-camera-formats-xenial-hwe.patch.txt](https://github.com/IntelRealSense/librealsense/files/5625051/realsense-camera-formats-xenial-hwe.patch.txt)

MartyG-RealSense, 2020-12-02:
Thanks very much @chia-labradorsystems for your advice to @TouchDeeper   :)

TouchDeeper, 2020-12-03:
@chia-labradorsystems Thanks for your advice, I use your patch, it work. But I found that the `v1-media-uvcvideo-mark-buffer-error-where-overflow.patch` also has problem, the patch this file want to apply is below:
```
diff --git a/drivers/media/usb/uvc/uvc_video.c b/drivers/media/usb/uvc/uvc_video.c
index fb86d6a..81a3530 100644
--- a/drivers/media/usb/uvc/uvc_video.c
+++ b/drivers/media/usb/uvc/uvc_video.c
@@ -1077,6 +1077,7 @@ static void uvc_video_decode_data(struct uvc_streaming *stream,
 	/* Complete the current frame if the buffer size was exceeded. */
 	if (len > maxlen) {
 		uvc_trace(UVC_TRACE_FRAME, "Frame complete (overflow).\n");
+		buf->error = 1;
 		buf->state = UVC_BUF_STATE_READY;
 	}
 }
```
But the line `buf->error = 1;` want to add already exist. I comment this patch then the `./scripts/patch-realsense-ubuntu-lts.sh` succeed.
The output of `sudo dmesg | tail -n 50` is:
```
[ 2232.450836] usbcore: deregistering interface driver uvcvideo
[ 2279.527160] media: Linux media interface: v0.10
[ 2279.530285] Linux video capture interface: v2.00
[ 2279.671020] uvcvideo: Found UVC 1.00 device Chicony USB2.0 Camera (04f2:b59e)
[ 2279.672379] uvcvideo 1-8:1.0: Entity type for entity Extension 4 was not initialized!
[ 2279.672382] uvcvideo 1-8:1.0: Entity type for entity Processing 2 was not initialized!
[ 2279.672385] uvcvideo 1-8:1.0: Entity type for entity Camera 1 was not initialized!
[ 2279.672550] input: Chicony USB2.0 Camera: Chicony  as /devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.0/input/input36
[ 2279.672780] usbcore: registered new interface driver uvcvideo
[ 2279.672782] USB Video Class driver (1.1.1)
```
The `registered new interface driver uvcvideo`  sholud indicate the script run succeed.

Then I run realsense-viewer, some warning occur:
```
 03/12 14:19:40,127 WARNING [139772010632960] (sensor.cpp:338) Unregistered Media formats : [ UYVY ]; Supported: [ ]
 03/12 14:19:45,192 WARNING [139772010632960] (types.cpp:57) set_pu(id=9) failed! Last Error: Input/output error
 03/12 14:19:45,192 WARNING [139772010632960] (sensor.cpp:722) Exception was thrown when inspecting properties of a sensor
 03/12 14:19:45,192 ERROR [139772010632960] (types.h:214) xioctl(VIDIOC_G_CTRL) failed Last Error: Invalid argument
 03/12 14:19:45,192 WARNING [139772010632960] (sensor.cpp:722) Exception was thrown when inspecting properties of a sensor
```
Does these waring affect?

TouchDeeper, 2020-12-03:
@MartyG-RealSense @chia-labradorsystems  I don't know why,  there will be new warnings occur now when I open realsense-viewer.

When I check on Stereo Module, a warning occurs:
`(ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to installation chapter for details`.
when I check on RGB Camera,  a warning occurs:
`UVC metadata payloads not available. Please refer to installation chapter for details.`
when I check on Motion Module, a warning occurs:
` HID timestamp not found! please apply HID patch`
I found that the hardware timestamp of all streams is not available except the accel stream.
[image]

TouchDeeper, 2020-12-03:
@MartyG-RealSense @chia-labradorsystems There will be new warning occurs like I post in my first comment. When I open the realsense-viewer,  warnings occur:
```
03/12 22:35:14,655 WARNING [140166347478784] (sensor.cpp:338) Unregistered Media formats : [ UYVY ]; Supported: [ ]
 03/12 22:35:14,664 WARNING [140166347478784] (backend-v4l2.cpp:1248) Pixel format 36315752-1a66-a242-9065-d01814a likely requires patch for fourcc code RW16!
 03/12 22:35:14,664 WARNING [140166347478784] (sensor.cpp:338) Unregistered Media formats : [ RW16 ]; Supported: [ ]
 03/12 22:35:19,711 WARNING [140166347478784] (types.cpp:57) set_pu(id=9) failed! Last Error: Input/output error
 03/12 22:35:19,712 WARNING [140166347478784] (sensor.cpp:722) Exception was thrown when inspecting properties of a sensor
 03/12 22:35:19,712 ERROR [140166347478784] (types.h:214) xioctl(VIDIOC_G_CTRL) failed Last Error: Invalid argument
 03/12 22:35:19,712 WARNING [140166347478784] (sensor.cpp:722) Exception was thrown when inspecting properties of a sensor
```
Then I check on `Stereo Module`, a warning occurs:
`(ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to installation chapter for details.`,

Then I check on `RGB Camera`, same waring occurs:
`(ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to installation chapter for details.`

Then I check on `Motion Module`, a warning occurs:
`(ds5-timestamp.cpp:174) HID timestamp not found! please apply HID patch.`

What's more, the hardware timestamp of all streams is not available except the accel stream.
[image]

Could you please provide some advice? Tanks a lot.

MartyG-RealSense, 2020-12-03:
@TouchDeeper   It looks as though hardware metadata support is not enabled on your librealsense build.  When performing a manual build, metadata support is provided by patching (a step you reported as having failed in the opening comment of this case).

If you are still using SDK **2.17.0** then could you try a **libuvc-backend** build of librealsense instead of a manual one if you have not tried it already, please?

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

Do not run the patching script after the libuvc-ackend build is completed, as patching is not necessary with the backend installation method and metadata support should be included in the build already.

TouchDeeper, 2020-12-04:
@MartyG-RealSense, @ev-mp   thank you for your reply.
I uninstall the installed SDK and reinstall according to `libuvc_installation.md`. It is worth noting that I just run a part of `libuvc_installation.md`:
```
sudo cp config/99-realsense-libusb.rules /etc/udev/rules.d/ 
sudo udevadm control --reload-rules && sudo udevadm trigger 
mkdir build && cd build
cmake ../ -DFORCE_LIBUVC=true -DCMAKE_BUILD_TYPE=release
make -j2
sudo make install
```
because I have installed the dependencies manually and I need the 2.17.1 version.

After installation, I run `rs-enumerate-devices` to verify the installation, but `No device detected. Is it plugged in?` occurs.

The information of `lsusb` is 
```
Bus 002 Device 008: ID 0dd8:0562 Netac Technology Co., Ltd 
Bus 002 Device 016: ID 8086:0b3a Intel Corp. 
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 04f2:b59e Chicony Electronics Co., Ltd 
Bus 001 Device 009: ID 24ae:1813  
Bus 001 Device 004: ID 8087:0aaa Intel Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

The related information of `usb-device` is:
```
T:  Bus=02 Lev=01 Prnt=01 Port=00 Cnt=01 Dev#= 16 Spd=5000 MxCh= 0
D:  Ver= 3.20 Cls=ef(misc ) Sub=02 Prot=01 MxPS= 9 #Cfgs=  1
P:  Vendor=8086 ProdID=0b3a Rev=50.c8
S:  Manufacturer=Intel(R) RealSense(TM) Depth Camera 435i
S:  Product=Intel(R) RealSense(TM) Depth Camera 435i
S:  SerialNumber=039223051536
C:  #Ifs= 6 Cfg#= 1 Atr=c0 MxPwr=440mA
I:  If#= 0 Alt= 0 #EPs= 1 Cls=0e(video) Sub=01 Prot=00 Driver=uvcvideo
I:  If#= 1 Alt= 0 #EPs= 1 Cls=0e(video) Sub=02 Prot=00 Driver=uvcvideo
I:  If#= 2 Alt= 0 #EPs= 1 Cls=0e(video) Sub=02 Prot=00 Driver=uvcvideo
I:  If#= 3 Alt= 0 #EPs= 0 Cls=0e(video) Sub=01 Prot=00 Driver=uvcvideo
I:  If#= 4 Alt= 0 #EPs= 1 Cls=0e(video) Sub=02 Prot=00 Driver=uvcvideo
I:  If#= 5 Alt= 0 #EPs= 1 Cls=03(HID  ) Sub=00 Prot=00 Driver=usbhid
```
The output of dmesg is:
```
[ 7040.975712] usb 2-1: new SuperSpeed USB device number 18 using xhci_hcd
[ 7040.996415] usb 2-1: New USB device found, idVendor=8086, idProduct=0b3a
[ 7040.996420] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 7040.996424] usb 2-1: Product: Intel(R) RealSense(TM) Depth Camera 435i
[ 7040.996427] usb 2-1: Manufacturer: Intel(R) RealSense(TM) Depth Camera 435i
[ 7040.996429] usb 2-1: SerialNumber: 039223051536
[ 7040.997799] uvcvideo: Unknown video format 00000050-0000-0010-8000-00aa00389b71
[ 7040.997807] uvcvideo: Unknown video format 00000032-0000-0010-8000-00aa00389b71
[ 7040.997898] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435i (8086:0b3a)
[ 7040.999580] uvcvideo: Unable to create debugfs 2-18 directory.
[ 7040.999772] uvcvideo 2-1:1.0: Entity type for entity Intel(R) RealSense(TM) Depth Ca was not initialized!
[ 7040.999776] uvcvideo 2-1:1.0: Entity type for entity Processing 2 was not initialized!
[ 7040.999780] uvcvideo 2-1:1.0: Entity type for entity Camera 1 was not initialized!
[ 7040.999989] input: Intel(R) RealSense(TM) Depth Ca as /devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1:1.0/input/input46
[ 7041.000759] uvcvideo: Unknown video format 36315752-1a66-a242-9065-d01814a8ef8a
[ 7041.000768] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435i (8086:0b3a)
[ 7041.001656] uvcvideo: Unable to create debugfs 2-18 directory.
[ 7041.001754] uvcvideo 2-1:1.3: Entity type for entity Processing 7 was not initialized!
[ 7041.001757] uvcvideo 2-1:1.3: Entity type for entity Extension 8 was not initialized!
[ 7041.001760] uvcvideo 2-1:1.3: Entity type for entity Camera 6 was not initialized!
[ 7041.111641] hid-sensor-hub 0003:8086:0B3A.0012: No report with id 0xffffffff found
[ 7041.186733] hid-sensor-hub 0003:8086:0B3A.0012: No report with id 0xffffffff found
```

MartyG-RealSense, 2020-12-04:
Mixing installations together that were created by different methods is not recommended.  Because **-DFORCE_LIBUVC=true** was used, it is likely that the dependencies were freshly installed automatically over the internet, potentially causing a conflict with dependencies that were already present from the previous 2.17.0 installation.

If you need to install the 2.17 manual installation, you should be able to by going to its **build** folder and inputting the instruction below:

**sudo make uninstall && make clean**

TouchDeeper, 2020-12-05:
@MartyG-RealSense I have done `sudo make uninstall && make clean`.
What's more, how can I specify the SDK version in the `libuvc_installation.sh`?

MartyG-RealSense, 2020-12-05:
Hi @TouchDeeper   You have a point about the problems of using **libuvc_installation.sh** for the installation, as it is set by default to look for the latest **master** version of the SDK instead of the **2.17** version that you want.   It may be possible to edit the path of the script to look for a local zip file containing the 17.1 source code but it is likely too much trouble to be worth trying.

Let's go back to the manual installation process.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md

Can you confirm please that when you perform the manual installation, you perform all of the steps before Step 3.

[image]

TouchDeeper, 2020-12-11:
@MartyG-RealSense 
I install the sdk2.40 according to the `libuvc_installation.sh`. 

I find that after I calibrated the IMU according to `IMU Calibration Tool for Intel® RealSense™ Depth Camera`,  if I reboot the PC, the calibration result will not work. I have to reload the calibration result by `python rs-imu-calibration.py -i accel_1.txt gyro_1.txt` to make it work.
Is this situation normal? Thank you a lot!

MartyG-RealSense, 2020-12-11:
Once the .txt file has been saved, the IMU calibration tool should then ask you if you want to save the calibration to the camera hardware.   Do you receive this message and input **Y** in response to it, please?

To quote the documentation:

***************************

Then the script will ask to save the data to the Device.
Choose Y (or any other combination with Y or y in it) to save.

That’s it. At the end a confirmation message appears:
`SUCCESS: saved calibration to camera.`

TouchDeeper, 2020-12-11:
Thank you for your reply. But I 'm sure I have entered Y when it ask me whether to save the result to camera.

MartyG-RealSense, 2020-12-11:
If you are still using SDK 2.17, I wonder if that could be a cause of the calibration not saving, as the SDK release notes say that the IMU calibration tool was first released in SDK 2.18.0, the next version after the one you are using.

https://github.com/IntelRealSense/librealsense/wiki/Release-Notes#release-2180

TouchDeeper, 2020-12-14:
@MartyG-RealSense 

> I install the sdk2.40 according to the `libuvc_installation.sh`.

The SDK version is 2.40. But it is strange that now the calibration result still works after I reboot.

MartyG-RealSense, 2020-12-14:
Thanks for the update @TouchDeeper - do you still have a problem that needs solving, please?

TouchDeeper, 2020-12-14:
@MartyG-RealSense I'm still testing, please give me more time. Thank you a lot.

MartyG-RealSense, 2020-12-14:
No problem, please update when you have test results.   Good luck!

TouchDeeper, 2020-12-14:
@MartyG-RealSense I have posted a new [issue](https://github.com/IntelRealSense/librealsense/issues/7988) about calibration, could you please have a look? Thank you a lot.

MartyG-RealSense, 2020-12-14:
I will look at it.   Thanks very much.

closed 2020-12-16 by TouchDeeper