realsenseai/librealsense #7864: Failed to connect to the hardware through USB 3.0 on MacOSX

ssghost, 2020-11-26:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D415 } | 
| Firmware Version                   | Not installed | 
| Operating System & Version |  MacOSX 10.15  | 
| Kernel Version (Linux Only)    |                                        | 
| Platform                                 | MacOSX 10.15  |
| SDK Version                            |  { 2.40.0 }                          | 
| Language                            |  {C++ }                          | 
| Segment			|  {others }                          | 

### Issue Description
I have an Intel Realsense D415 camera connected through type-c to USB 3.1 line to my MacbookPro. Librealsense SDK 2.40.0 has been successfully installed on this Mac. I also installed libusb-1.0.0 and lsusb through Homebrew on this machine. 

The Infrared light generator is on, I can see red striped light shining inside it when connected to the USB line.

But all librealsense SDK terminal tools can't recognize this device, except lsusb which can find this device and print its serial number.

Running ``lsusb | grep 8086`` and ``rs-fw-update -l`` on Mac Terminal and the results are like below:
[image]

ssghost, 2020-11-26:
I suddenly solved this issue by remove all USB 3.0 connected hardware and leave only D415 camera connected through USB 3.0 to my Macbook Pro.

This means librealsense SDK 2.40.0 works only when there is nothing else but the camera is occupying the USB port.

closed 2020-11-26 by ssghost

Starslayerx, 2020-11-30:
I got same issue. And sloved same way.
Do you know what caused this? I wanna to connect my mouse and  keyboard. @ssghost