realsenseai/librealsense #7865: Why the translation results of y and z axis between IMU and camera are reverse with RealSenseD435i?

opendvd, 2020-11-27:
I use Kailbr to calibrate the RealSenseD435i, especially the external parameters between IMU and camera. But the external parameters of translation between IMU and camera confused me. The Kalibr's results are different with ideal value.
The ideal value is:
t_IC0 = [-0.006 0.005 0.012]T;
t_IC1 = [0.044 0.005 0.012]T;
But kalibr result is:
t_IC0 = [-0.00510796 0.01859231 0.00576145]T;
t_IC1 = [0.04383316 0.01886102 0.00574488]T;
So why the results of y and z axis are reverse?
Kalibr's result is here: [results-imucam-imu_cameras_calibration.txt](https://github.com/IntelRealSense/librealsense/files/5607870/results-imucam-imu_cameras_calibration.txt)

MartyG-RealSense, 2020-11-27:
Hi @opendvd  If you are seeking to understand the transformation matrices between the IMU and the D435i's other camera sensors then the information in the link below may be helpful to you.

https://github.com/IntelRealSense/realsense-ros/issues/1449

opendvd, 2020-11-29:
> Hi @opendvd If you are seeking to understand the transformation matrices between the IMU and the D435i's other camera sensors then the information in the link below may be helpful to you.
> 
> [IntelRealSense/realsense-ros#1449](https://github.com/IntelRealSense/realsense-ros/issues/1449)

Thank you. I checked this hyperlink that you provided, but did not find the answer I wanted. I want to know what caused the result of translation to be reverse?
I disassembled the realsenseD435i and used a ruler to measure, the result is very close to the ideal value:
t_IC0 = [-0.006 0.005 0.012]T;
t_IC1 = [0.044 0.005 0.012]T;
(The ideal value is got by realsense2_description ROS wrapper. I run the launch file: view_d435i_model.launch in ROS. )
But when I use kalibr to get the calibration results, the results is reverse. I also use kalibr to calibration the MYNTEYE camera, its calibration is right, so I think Kalibr is correct. The problem is in realsense.

opendvd, 2020-11-29:
The transformation about the RealsenseD435i is show in rviz.
[image]

MartyG-RealSense, 2020-11-29:
Is Kalibr takng account of how the left infrared imager is on the **right** side of the camera when viewed from a front-on perspective?

https://github.com/IntelRealSense/librealsense/issues/3662

opendvd, 2020-11-30:
> Is Kalibr taking account of how the left infrared imager is on the **right** side of the camera when viewed from a front-on perspective?
> 
> #3662

1. Thank you for your reply. The Kalibr only focus on camera's ros topic: `/camera/infra1/image_rect_raw`&`/camera/infra2/image_rect_raw`. 
2. I think it has nothing to do with Kalibr, because I use Kalibr to calibrate my MYNTEYE camera is normal.
3. Could you please calibrate D435i with Kalibr? (The transformation between IMU and two camera) You will understand my problem.

MartyG-RealSense, 2020-11-30:
I am not able to make use of Kalibr.

It has already been established in another case that there can be large differences in Kalibr's calibration results with RealSense and with Mynteye.

https://github.com/IntelRealSense/librealsense/issues/7103

Bearing that in mind, I do not think that there is anything else that I can add that will be helpful, unfortunately.  I would always recommend using RealSense's own IMU calibration tool.

opendvd, 2020-12-01:
> I am not able to make use of Kalibr.
>
> It has already been established in another case that there can be large differences in Kalibr's calibration results with RealSense and with Mynteye.
>
> #7103 [...]

Thanks for your reply. I will continue to explore the cause of this problem.

MartyG-RealSense, 2020-12-07:
Hi @opendvd   Do you have an update that you can provide, please?   Thanks!

closed 2020-12-21 by MartyG-RealSense