realsenseai/librealsense #7866: Transforming camera coordinates to world coordinates

AlexandrineRibeiro, 2020-11-27:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435  | 
| Operating System & Version |   Win 10  | 
| Language                            |  python                        | 
                     

### Issue Description
I want to find the extrinsics of my cameras related to a pre-defined real-word coordinated system. 

Would using the following function help achieve this:
1. Calibrate camera1 and camera2 extrinsic relative to reach other from a common plan in the real-world (How?)
2. For camera1 (x,y,z) and extrinsic of camera1 calculated in (1.) get from_point
3. For camera2 (x,y,z) and extrinsic of camera2 calculated in (1.) get from_point

Both points should be same in the real world.

It would be really helpful if you could share your experience.

These issues tried to get answers to the same problem but without success
https://github.com/IntelRealSense/librealsense/issues/2664
https://github.com/IntelRealSense/librealsense/issues/1904

Thanks

MartyG-RealSense, 2020-11-27:
Hi @AlexandrineRibeiro   What you are aiming to achieve essentially sounds like what the RealSense Python example program **box_dimensioner_multicam** does.  "This sample demonstrates the ability to use the SDK for aligning multiple devices to a unified co-ordinate system in world".

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python/examples/box_dimensioner_multicam

If you are open to using ROS, it may also be worth looking at Intel's tutorial for capturing a pair of point clouds woth 2 cameras at different angles and publishing the spatial connection between the cameras via a Python script to stitch the clouds together into one combined cloud.

https://github.com/IntelRealSense/realsense-ros/wiki/Showcase-of-using-2-cameras

AlexandrineRibeiro, 2020-11-30:
Thank you for the indications. I have been studying example **box_dimensioner_multicam**.

Can you tell me where the coordinate system of the chessboard itself is located?

MartyG-RealSense, 2020-11-30:
It looks as though the world coordinates of the chessboard's pose (position and angle) are calculated at this line of the script:

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/box_dimensioner_multicam/box_dimensioner_multicam_demo.py#L67

MartyG-RealSense, 2020-12-07:
Hi @AlexandrineRibeiro   Do you require further assistance with this case, please?  Thanks!

AlexandrineRibeiro, 2020-12-07:
Hi @MartyG-RealSense, I don't need more assistance with this. I was able to do what I intended based on the [box_dimensioner_multicam](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python/examples/box_dimensioner_multicam) example.

Feel free to close this issue.

Alexandrine

MartyG-RealSense, 2020-12-07:
Excellent news @AlexandrineRibeiro - thanks for the update!

closed 2020-12-07 by MartyG-RealSense