realsenseai/librealsense #7868: Install RS300 on Pop!_Os?

Starslayerx, 2020-11-27:
Hello, I'm trying to install rs300 on my pop_os linux(ubuntu based).
- I tried to install the ubuntu version but failed.
- And I also tried a docker container found in issue. But I still got problems because of my nvidia gpu.
Do you have solutions for other linux distributions?
(I don't wanna to use VirtualMachines, it's slow.)

MartyG-RealSense, 2020-11-27:
Hi @Starslayerx   You can build librealsense from source code with CMake using a method called **RSUSB** that is not dependent on Liux versions or kernel versions and does not require patching.   It does require an internet connection for the installation process though.  It involves adding the following customization flag to the CMake build instruction:

**-DFORCE_RSUSB_BACKEND=true**

https://github.com/IntelRealSense/librealsense/issues/6368#issuecomment-636316279

CUDA support to make use of your Nvidia GPU can be enabled in the buikd by adding this term to the build instruction:

**-DBUILD_WITH_CUDA:bool=true**

Starslayerx, 2020-11-27:
Thanks, I think I have installed  successfully.

closed 2020-11-27 by Starslayerx

MartyG-RealSense, 2020-11-27:
Great news - thanks for the update!   :)