realsenseai/librealsense #7870: Measuring USB bandwidth: theory vs measurements

jurc192, 2020-11-28:
I have a question regarding USB bandwidth of a single RealSense (D435i) camera, since measuring it (with `usbtop`) gave confusing numbers which don't match my estimations at all.

Single camera, streaming with RGB8, 848x480 px, 6 fps, connected via USB 3 interface should use approximately 59 Mbps (59 * 10^6 bits per second) of bandwidth. This was calculated using the following formula:
```
n_pixels * bits_per_pixel * framerate = bandwidth (bps)
(848*480) px * 24 bpp * 6 fps = 58613760 bpp ~ 59 Mbps
```

1. Is it true that RS2_FORMAT_RGB8 equals to 24 bits per pixel?
2. Is this data internally converted for transport, which might be the reason for the offset between my estimations and measurements? What kind of pixel format is used in USB 'packets'? 
3. What kind of bandwidth consumption should I expect in my case?

Thanks!

MartyG-RealSense, 2020-11-29:
Hi @jurc192 

1.  A list of the bits returned by each image format can be found in the librealsense file **image.cpp**.  It states that RGB8 returns 24 bits.  

https://github.com/IntelRealSense/librealsense/blob/master/src/image.cpp#L30

2.  I will refer this question to @dorodnic the RealSense SDK Manager to ensure that you receive a correct explanation.

3.  I believe that the structure of your bandwidth calculation formula is basically correct:  

**(resolution x bit depth) * FPS**

I do not have figures for 6 FPS, though Intel produced bandwidth charts for their multiple camera white paper that provide bandwidth estimates in various modes for one camera or more at 30+ FPS.  They tested with both direct to computer connections and USB hub.

https://dev.intelrealsense.com/docs/multiple-depth-cameras-configuration#section-2-multi-camera-considerations

[image]

jurc192, 2020-11-29:
Thanks for your answer! 
Assuming that the pixel formats used in the first row of the chart above are Z16 (depth) and RGB8 (color),  the total bandwidth should be 1750 Mbps which is not the case. So my assumption is probably incorrect and the color stream is probably transported with 16-bit data - which results in 1172 Mbps bandwidth indeed. 

Is color data always transported using YUYV format (16-bit per pixel) no matter what  you select in rsViewer/pipeline API?
There is no clear documentation about this conversion, so I was pretty confused when making calculations. [This document](https://github.com/IntelRealSense/librealsense/blob/master/doc/frame_lifetime.md#frame-memory-management) kinda touches the subject, but it does not clearly mention that the data is converted during transport.

MartyG-RealSense, 2020-11-29:
The data sheet for the 400 Series documents states that the color stream from the RGB sensor is 16-bit YUY2, whilst the color stream from the left IR imager is 16-bit UYVY.

[image]

In firmware **5.12.8.200** support on the D455 model for a color stream from the left IR sensor was added.

[image]

[image]

jurc192, 2020-11-29:
Ahaa yes, now it makes sense. Thanks! :)

MartyG-RealSense, 2020-12-02:
Regarding your latest case on https://github.com/IntelRealSense/librealsense/issues/7904 - that is a set of questions that one of the developers would be best equipped to answer.  I wanted to offer some supporting material outside of the case though.  There is an hour long YouTube seminar about the librealsense2 software architecture presented by Dorodnic the RealSense SDK Manager.  The architecture details start at 22 minutes 30 seconds into the video.

https://m.youtube.com/watch?v=uz2dGbjA3j4

closed 2020-12-16 by MartyG-RealSense