realsenseai/librealsense #7871: D455 freeze of RGB Camera stream when gyro/accel are enabled

robbel, 2020-11-29:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | 05.12.09.00 | 
| Operating System & Version |   Win 10 | 
| Platform                                 | PC |
| SDK Version                            |  2.40.0 | 

### Issue Description
As soon as I enable the "Motion Module" stream in the RealSense Viewer, the RGB Camera stream freezes and a "No Frames Received!" warning is shown after a couple of seconds. In the debug panel, Frame Drops per Second goes up to ~100%. The error/warning output shows: 
> synthetic-stream.cpp:48 - Exception thrown during user processing callback: Error occurred during execution of the processing block

and
> rs.cpp:1091 - null pointer passed for argument "frame"

This is independent of which motion module is enabled (gyro, accel, or both).

MartyG-RealSense, 2020-11-29:
Hi @robbel   This is a known issue and is not specific to the D455 model.  You are correct that the issue can occur no matter which order the streams are activated in.  The trigger is when the Motion Module is sctivated.

There are workarounds to enable using depth, RGB and Motion Module simultaneously when programming your own application in C++ or Python languages but there is not a solution currently available for the RealSense Viewer program.

You can access the multiple streams simultaneously without the RGB **No Frames Received** problem occurring if you use the **rs-multicam** or the **rs-capture** example programs, though the data cannot be interacted with.  If you have installed the full Windows version of the RealSense SDK, you can find these programs by right-clicking on the Viewer's launch icon on the Windows desktop and selecting the menu option **Open file location**.

robbel, 2020-11-29:
Thanks @MartyG-RealSense

MartyG-RealSense, 2020-12-05:
Hi @robbel   Do you require further assistance with this case, please?   Thanks!

robbel, 2020-12-05:
@MartyG-RealSense not at this time, I'll wait for a firmware or viewer update. Thank you

MartyG-RealSense, 2020-12-05:
Thanks very much for the update and for your patience @robbel

closed 2020-12-12 by MartyG-RealSense