realsenseai/librealsense #7873: Does SR300 support opencv3 or 4?

Starslayerx, 2020-11-29:
Does SR300 support opencv3 or 4? 
I don't know where to find the doc,thanks.

MartyG-RealSense, 2020-11-29:
Hi @Starslayerx   If a RealSense program that can work with SR300 makes use of OpenCV then I would recommend assuming that OpenCV 3 is the minimum requirement unless the instructions state otherwise (for example, stating a specific OpenCV 3.x version or stating that OpenCV 4.x is required).

For example, the librealsense OpenCV wrapper page states that the OpenCV sample programs require minor code changes to work with OpenCV 4.x.  You can interpret from this that they should work without changes with OpenCV 3..x

https://dev.intelrealsense.com/docs/opencv-wrapper

Starslayerx, 2020-11-30:
Thx.
I got another issue same as this [issue7864](https://github.com/IntelRealSense/librealsense/issues/7864), but my system is ubuntu20 lts.How can I fix this?

Starslayerx, 2020-11-30:
```cpp
$ ls -ltrh /dev/video*
crw-rw----+ 1 root video 81, 0 11月 30 08:58 /dev/video0
crw-rw----+ 1 root video 81, 1 11月 30 08:58 /dev/video1
```
Not find SR300(whatever I plug only SR300), but I can find it with only SR300 plugged by c code [here](https://github.com/IntelRealSense/librealsense/blob/master/examples/C/distance/rs-distance.c)

Starslayerx, 2020-11-30:
I think it's sloved by itself some how...

closed 2020-11-30 by Starslayerx

MartyG-RealSense, 2020-11-30:
I'm pleased to hear that the problem got resolved - thanks very much for the update!