realsenseai/librealsense #7874: HDR parameter with python

Fred3D-tech, 2020-11-29:
Question...
using advanced parameters (advanced_mode = rs.rs400_advanced_mode(device)) I'm trying to find the command to set Hdr=on.
I guess it is something like depth_sensor.set_option(rs.option.HDR)... but I can't find the right command..
Thanks for the answer !

MartyG-RealSense, 2020-11-29:
Hi @Fred3D-tech  The RS2 version of HDR instructions - such as **RS2_OPTION_HDR_ENABLED** - can be found in the link below.  That should hopefully provide a reference for conversion to Python.

https://intelrealsense.github.io/librealsense/doxygen/rs__option_8h.html#a8b9c011f705cfab20c7eaaa7a26040e2a2f4b24b7d109582742c4a9168fbdef0e

The **2.39.0** release notes relating to HDR also state that a white paper on the subject will be coming that will provide in-depth presentation and usage guidance.

https://github.com/IntelRealSense/librealsense/wiki/Release-Notes#release-2390

I am off-shift now but if you leave a comment below then I will be pleased to respond when I return on Monday.

Fred3D-tech, 2020-11-30:
[image]

I had a look at the document and I used this ref https://community.intel.com/t5/Items-with-no-label/Enable-Auto-Exposure-Using-Python-Code/td-p/642319 to see if the HDR_Enabled option could be set. But as you can see, the parameter is not available (or exposed) through the python API.
Is there a solution to solve that?

dorodnic, 2020-11-30:
Please double check pyrealsense2 version - 2.39 should include all HDR related capabilities

Fred3D-tech, 2020-12-01:
[image]
I have now the 2.40 installed (and it was difficult to make it work!) and same problem: nothing related to HDR
[image]

MartyG-RealSense, 2020-12-01:
Hi @Fred3D-tech   What is the firmware version installed in your camera, please?  The API change notes for SDK 2.39.0 (which HDR was introduced in) state that the **RS2_OPTION_HDR_ENABLED** instruction requires firmware **5.12.8.200**.

https://github.com/IntelRealSense/librealsense/wiki/API-Changes#version-2390

Fred3D-tech, 2020-12-01:
I installed the last firmware... the hrd option is working very well on Intel realsense Viewer. It is not working with the python api.
[image]

MartyG-RealSense, 2020-12-02:
Hi @Fred3D-tech  The white-paper document about HDR is now available from the link below.

https://dev.intelrealsense.com/docs/high-dynamic-range-with-stereoscopic-depth-cameras

Python code for enabling HDR with **rs.option.hdr_enabled** can be found in this section of the paper:

https://dev.intelrealsense.com/docs/high-dynamic-range-with-stereoscopic-depth-cameras#section-3-2-controlling-hdr-programmatically

Fred3D-tech, 2020-12-08:
thanks...
Here is a piece of code working:
import pyrealsense2.pyrealsense2 as rs

pipeline = rs.pipeline()
config = rs.config()
profile = pipeline.start(config) # Start streaming
sensor_dep = profile.get_device().first_depth_sensor()
print ("Trying to set Exposure")
exp = sensor_dep.get_option(rs.option.exposure)
print ("exposure = %d" % exp)
print ("Setting exposure to new value")
exp = sensor_dep.set_option(rs.option.exposure, 25000)
exp = sensor_dep.get_option(rs.option.exposure)
print ("New exposure = %d" % exp)
sensor_dep.set_option(rs.option.hdr_enabled, True)
 
profile = pipeline.stop

MartyG-RealSense, 2020-12-08:
Thanks very much @Fred3D-tech for sharing with the RealSense community the solution that works for you   :)

closed 2020-12-21 by MartyG-RealSense