realsenseai/librealsense #7875: Realsense viewer can't find realsenseD455 camera on Jetson AGX Xavier

Mingchi-Yeh, 2020-11-30:
Realsense viewer can't find realsenseD455 camera on Jetson AGX Xavier (L4T 32.3.1)

Librealsense : v2.39.0

RealSense camera(s): D455

When I install the librealsense v2.39.0 ,I turn on the Realsense Viewer and used the D455 camera , but realsense viewer can't found the D455.
But when I tried D435 , the realsense viewer work fine.
Is D455 not currently supported? Or your company have any technical support?

RealSenseSupport, 2020-11-30:
Hi @Mingchi-Yeh 
Is the  D455 is enumerated , e.g use lsusb command to see if D455 is listed ?

Mingchi-Yeh, 2020-11-30:
The lsusb can find the "Bus 002 Device 003: ID 8086:0b5c Intel Corp. ".
But realsense viewer still can't find the D455 camera.

RealSenseCustomerSupport, 2020-12-01:
----------------------------------------------

To use the D455 please make sure to,
1) use the latest 2.40.0 LRS,
2) use the procedure for either building from source using RSUSB Backend here, or building from source using Native Backend here.
 3) run the script to patch the kernel ( ./scripts/patch-realsense-ubuntu-L4T.sh )

RealSenseCustomerSupport, 2020-12-04:
----------------------------------------------

Hello @Mingchi-Yeh,

Have you been able to get the D455 camera to connect on the Jetson AGX platform? Let me know if you are still needing assistance, if you do not need further help we will close the ticket in seven days.

Regards,
Ted Tsuchiya.

Mingchi-Yeh, 2020-12-05:
HI @RealSenseCustomerSupport 
The D455 is works fine!
Thanks for your advice!

RealSenseCustomerSupport, 2020-12-07:
----------------------------------------------

@Mingchi-Yeh, great to hear that your D455 is now working! I will close this issue but please feel free to open another issue for any new problems you may encounter.

Best regards,
Ted Tsuchiya.

closed 2020-12-07 by RealSenseCustomerSupport