realsenseai/librealsense #7883: Feature Request: RealSense D455 support inn the Unity wrapper for Unity 2019 / 2020

MartyG-RealSense, 2020-11-30:
This message is a Feature Request for adding RealSense D455 support in the Unity wrapper for the 2019 / 2020 versions of Unity.  Currently the 2017 versions of Unity support D455.

This request is supported by RealSense users in the case links below:

https://github.com/IntelRealSense/realsense-ros/issues/1420
https://github.com/IntelRealSense/librealsense/issues/7662

This case should be kept open whilst the Feature Request is open.

wodndb, 2021-01-25:
Have you any plan for supporting D455 in Unity?

MartyG-RealSense, 2021-01-25:
Hi @wodndb  As mentioned above, D455 is supported in the old **Unity 2017** and there is an official Intel feature request filed to add D455 support to the RealSense Unity wrapper.  

https://github.com/IntelRealSense/librealsense/issues/7662#issuecomment-718098591

I do not have any information to provide about the current status of that feature request though.

MartyG-RealSense, 2021-02-02:
Hi again @wodndb  Intel have stated about this feature request that D455 support in Unity was provided in SDK **2.40.0**.  In that SDK version, Intel provided an update to the Unity wrapper to adapt it for Unity 2018 to 2020.

https://github.com/IntelRealSense/librealsense/pull/7625

closed 2021-02-02 by MartyG-RealSense