realsenseai/librealsense #7886: Exporting Point Cloud in C#

BJuraj, 2020-11-30:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   { Win 10 }  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | { PC } |
| SDK Version                            |  { 2.40.0 }                          | 
| Language                            |  {C# }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description

Hello I'm trying to export a point cloud in C#. I'm using the point cloud code from the cookbook and the exporttoply() method, but when I run the code nothing is exported. 
Hopefully someone can tell me how to achieve this.
Thanks in advance.

`              
                var pipe = new Pipeline();
                var pc = new PointCloud();
                pipe.Start();

                
                    using (var frames = pipe.WaitForFrames())
                    using (var depth = frames.DepthFrame)
                    using (var color = frames.ColorFrame)
                    using (var points = pc.Process(depth).As<Points>())
                    {
                       /* Console.WriteLine("The camera is pointing at an object " +
                            depth.GetDistance(depth.Width / 2, depth.Height / 2) + " meters away\t");

                        Console.SetCursorPosition(0, 0);*/

                        // CopyVertices is extensible, any of these will do:
                        var vertices = new float[points.Count * 3];
                        // var vertices = new Intel.RealSense.Math.Vertex[points.Count];
                        // var vertices = new UnityEngine.Vector3[points.Count];
                        // var vertices = new System.Numerics.Vector3[points.Count]; // SIMD
                        // var vertices = new GlmSharp.vec3[points.Count];
                        //  var vertices = new byte[points.Count * 3 * sizeof(float)];
                        points.CopyVertices(vertices);

                                                                 
                       points.ExportToPLY("pointcloud.ply", color);

                   // Points myPoints = new Points();
                   // myPoints.ExportToPLY("pointcloud.ply", color);

                    }
`

MartyG-RealSense, 2020-11-30:
Hi @BJuraj   Have you seen the adaptation of the C# cookbook point cloud for **points.ExportToPly** that is in the link below, please?

https://github.com/IntelRealSense/librealsense/issues/7362

BJuraj, 2020-11-30:
@MartyG-RealSense Thanks for the fast reply. The code from your link solved my problem.

MartyG-RealSense, 2020-11-30:
Great to hear - thanks for the update   :)

BJuraj, 2020-11-30:
@MartyG-RealSense Sorry to reopen this issue again. Is it possible to get the point cloud without meshing?

MartyG-RealSense, 2020-11-30:
I researched your follow-up question deeply.  In C++ and Python you can set ply configuration options wit the **save_to_ply** processing block but I could not find a C# method for accessing it.

https://github.com/IntelRealSense/librealsense/pull/4906

BJuraj, 2020-11-30:
Ok. Thanks for your help.

closed 2020-11-30 by BJuraj