realsenseai/librealsense #7889: Update to latest ImGui

yanickrochon, 2020-11-30:
The examples aren't up-to-date with the latest changes of [ImGui](https://github.com/ocornut/imgui).

dorodnic, 2020-11-30:
Why? Is the new version of IMGUI not working with some of the examples? 
We have fixed specific version that worked well for us, and I prefer to keep it that way, unless there are good reasons to upgrade.

yanickrochon, 2020-11-30:
Hi @dorodnic , I was trying the post processing example, and this line failed :

https://github.com/IntelRealSense/librealsense/blob/34fc284d537a4b873a37b737a61f1f1da92dbb60/examples/post-processing/rs-post-processing.cpp#L254

Other examples I tried run fine, even if I use `make` instead of `cmake`. (I did copy `example.hpp` as well), so everything's fine, until I got the the post processing one. So far, I had copy pasted examples directly into my project, and so I linked ImGUI directly from their repository so I could update my project when they would update theirs. However, after some time of trying to fix the example, nothing worked. I got errors and segfaults and so I opened this issue.

Of course, this is not a bug, but I would assume you guys would link to their git as a sub-module instead of copying their files over and fall behind. It would help those like me understand how things work without receiving errors like these.

RealSenseSupport, 2020-12-30:
@yanickrochon Thanks very much for your input!  As @dorodnic replied, currently there's no plan to update ImGui yet. Will close the ticket accordingly. Thanks!

closed 2020-12-30 by RealSenseSupport