realsenseai/librealsense #7890: A question regarding licenses

neilyoung, 2020-11-30:
This question is probably very easy to answer: Given I would build a device with a T265 and my software on that device: Would I have to consider to pay for some licenses to Intel or are there any GPL licenses to be considered?

Am I free to sell source code and device "license free", using the camera and utilizing the IntelRealsense SDK?

dorodnic, 2020-11-30:
Hi @neilyoung 
The SDK is covered in its entirety by [Apache 2](https://github.com/IntelRealSense/librealsense/blob/master/LICENSE) license, giving you the right to use it commercially in your software. No part of the SDK (including the firmware images) contain GPL code.

neilyoung, 2020-11-30:
Thank you very much, Sergey, for this quick and profound answer.

closed 2020-11-30 by neilyoung