realsenseai/librealsense #7891: building from source on windows 10

Fred3D-tech, 2020-11-30:
trying to compile from source on windows 10..
I used this document: https://dev.intelrealsense.com/docs/compiling-librealsense-for-windows-guide
do you know that this is not working? honestly, are you serious? can you give a simple and working solution for windows 10?I can't believe that a company like Intel is not able to get a serious documentation. 
Ok... just give us a serious and working solution to build from source on windows 10.

Fred3D-tech, 2020-11-30:
by the way, pip install pyrealsense2 install (2.36.0.2038) and not 2.40 as expected !

MartyG-RealSense, 2020-12-01:
Hi @Fred3D-tech   I apologise that you have had a frustrating experience with installation of the SDK from source on Windows 10.

To simplify the instructions, you could skip the majority of the guide about enabling metadata.  This is because after that guide was published, an easier method to enable metadata support through the RealSense Viewer program was created.  So I would reommend just focusing on the CMake instructions in the opening section of the guide.

Are you having problems with the CMake process too, please?  If you are, please provide details.  Thanks!

closed 2020-12-15 by MartyG-RealSense