realsenseai/librealsense #7893: Update Realsense d415 firmware through bash script

pmirabel, 2020-11-30:
Hi, 

I'm using a D415 device attached (USB type: 3.2) to a Debian stretch board. 
If i `rs-fw-update -f /data/fw/rs/D4XX_FW_Image-5.12.9.0.bin` in a terminal,  I can see ` 30/11 20:19:32,715 ERROR [140478469534464] (types.h:313) get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device` but right after, upgrade continue with no error.

if I do it through a bash script when attached device is not up to date I get a :
`nothing to do, run again with -h for help`  

And when device is up to date : 
`update to FW: /(...)/D4XX_FW_Image-5.12.9.0.bin` but that's all, script stops here.

Any advice ?

MartyG-RealSense, 2020-12-01:
Hi @pmirabel   Using RealSense Viewer  or rs-fw-update to update firmware is recommended.  If you do need to incorporate firmware update into your application as a script though, the discussion in the link below provides programming advice.

https://github.com/IntelRealSense/librealsense/issues/4416

pmirabel, 2020-12-01:
> Hi @pmirabel Using RealSense Viewer or rs-fw-update to update firmware is recommended. If you do need to incorporate firmware update into your application as a script though, the discussion in the link below provides programming advice.
> 
> #4416
Thank you, very much, I will take some time to figure out if I really need to script this. My original issue was caused by messed up installation of my system! sorry about that.

> 
> If you would prefer to use rs-fw-update as you did in your original attempt and you encounter **No such device**, is that message followed by a **RSB_USB_STATUS_ACCESS** message? If it is, that may indicate a permissions problem.
> 
> #7521
This message continue to pop once each time I execute the rs-fw-update tool. 
> [140478469534464] (types.h:313) get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device   

Can it be somehow related to kernel? Do you now if there is any way to see more debug message?

MartyG-RealSense, 2020-12-01:
Any time that there is an error that contains **ioctl(UVCIOC_CTRL_QUERY)**, I would  suspect a kernel conflict first.

You can set the error logging level of librealsense.

https://dev.intelrealsense.com/docs/troubleshooting#section-enable-lib-real-sense-log

https://dev.intelrealsense.com/docs/librealsense-error-handling-scheme#section-errors-warnings-and-exceptions

anhTuanBrycen, 2020-12-03:
> Any time that there is an error that contains **ioctl(UVCIOC_CTRL_QUERY)**, I would suspect a kernel conflict first.
>
> You can set the error logging level of librealsense. [...]

Hello, as far as I know, you are a member of the realsense team. Please solve this problem. https://github.com/IntelRealSense/librealsense/issues/7902

MartyG-RealSense, 2020-12-03:
Hi @anhTuanBrycen   I am a 400 Series and SR30x specialist.  I am not able to handle L515 and T265 cases, as they are the responsibility of other RealSense team members.

anhTuan0712, 2020-12-03:
> Hi @anhTuanBrycen I am a 400 Series and SR30x specialist. I am not able to handle L515 and T265 cases, as they are the responsibility of other RealSense team members.

Hi MartyG-RealSense ,Thank you for your feedback,!

MartyG-RealSense, 2020-12-10:
Hi @pmirabel   Do you require further assistance with this case, please?

closed 2020-12-17 by MartyG-RealSense