realsenseai/librealsense #7894: No frames recevied

qingweihk, 2020-12-01:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435i } | D435i 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 05.12.09.00
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | ubuntu20.04
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |PC
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 2
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
I open Stereo module and RGB camera(on ubuntu20.04),but got no frames received while it is ok   
very little times.But on the same PC,I  try on ubuntu18.04 and win10 ，when open relsense-viewer,the show is good and no error like ''no frames recevied"
Any help?

MartyG-RealSense, 2020-12-01:
Hi @qingweihk   If you are experiencing frame drops or No Frames Received on Linux, the SDK release notes advise installing librealsense with a method called **RSUSB backend**.

https://github.com/IntelRealSense/librealsense/wiki/Release-Notes#release-2400

The RSUSB installation method requires an internet connection to perform the installation but is not dependent on inux versions or kernel versions and does not need to be patched.  An RSUSB build is enabled by building from source code with CMake and adding the flag **-DFORCE_RSUSB_BACKEND=true** to the CMake build instruction.

https://github.com/IntelRealSense/librealsense/issues/6368#issuecomment-636316279

MartyG-RealSense, 2020-12-14:
Case clsoed due to no further comments received.

closed 2020-12-14 by MartyG-RealSense