realsenseai/librealsense #7896: Screen inverted when D455 connected with my Ubuntu

Steven-ZY, 2020-12-01:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455| 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Linux (Ubuntu 18.04)  | 
| Kernel Version (Linux Only)    |  5.4.0                                       | 
| Platform                                 | PC |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

MartyG-RealSense, 2020-12-01:
Hi @Steven-ZY   It is usually caused by RealSense cameras that have an internal IMU triggering the screen rotation controls in Ubuntu (the one that rotates the view on a device equipped with a gyro such as a tablet).  You can fix the problem by turning off screen rotation in Ubuntu.

https://github.com/IntelRealSense/librealsense/issues/6941

Steven-ZY, 2020-12-01:
@MartyG-RealSense Thank you very much!

MartyG-RealSense, 2020-12-01:
You are very welcome   :)

MartyG-RealSense, 2020-12-08:
Case closed due to solution provided and no further comments received.

closed 2020-12-08 by MartyG-RealSense