realsenseai/librealsense #7897: Left IR-Imager RGB8

BJuraj, 2020-12-01:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (10) }  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  {  2.40.0}                          | 
| Language                            |  {C++ }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

Hello I want to save an image of the left IR-imager in an RGB8 format, but my code always crashes when it reaches pipe.start(cfg);.
If I use RS2_FORMAT_Y8 it works fine, but i would like to have an color image of the IR-imager.
Am I doing something wrong or is this not possible?

`	
rs2::config cfg;
cfg.enable_stream(RS2_STREAM_INFRARED, 1, 1280, 720, RS2_FORMAT_RGB8);

pipe.start(cfg);
`

MartyG-RealSense, 2020-12-01:
Hi @BJuraj   Are you receiving a **Could not resolve request** error please?  This could indicate that you are trying to access the color stream from the left IR imager on a RealSense model that doesn't support it.   You can stream RGB8 from a D415 or from a D455 that has firmware **5.12.8.200** or newer.  On the D435 and D435i models, both left and right IR are Y8 monochrome.

**D415**

[image]

**D455**

[image]

**D435 / D435i**

[image]

BJuraj, 2020-12-01:
The error I'm getting is "RealSense error calling rs2_pipeline_start_with_config(pipe:03482960, config:051108B0): Couldn't resolve requests"

I'm using a d415 with the newest firmware. Sorry I forgot to mention that.

MartyG-RealSense, 2020-12-01:
On 1280x720, the maximum FPS for the left IR color stream is 30, whilst on 848x480 and below it is 90 FPS.  So can you try adding the FPS to the end of the bracket please to make sure that the FPS is correctly defined for that resolution:

**cfg.enable_stream(RS2_STREAM_INFRARED, 1, 1280, 720, RS2_FORMAT_RGB8, 30);**

[image]

MartyG-RealSense, 2020-12-14:
Case clsoed due to no further comments received.

closed 2020-12-14 by MartyG-RealSense