realsenseai/librealsense #7899: failed to set power state / Jetson AGX Xavier

Fred3D-tech, 2020-12-01:
Realsense version 2.40
Jetson AGX Xavier : updated to the last version available
1 x Dr415 connected on a AmazonBasics Hub USB-C 3.1 1 ports with a power supply 36W (12V/3A)

Error message: failed to set power state

  File "/home/fred/scripts/sensors/CaptureX_Realsense.py", line 1456, in <module>
    depth_table = advanced_mode.get_depth_table()
I don't understand.. it is working on the viewer well, and the power supply should eliminate this kind of problem. Why is it not working?
I have 18 DR415, 8 DR450 and 6 D515.. I use Intel nucs, jetson nano and Xavier and a very powerfull PC: I can't get a stable working configuration. Please solve your issues before making new versions !

MartyG-RealSense, 2020-12-02:
Hi @Fred3D-tech   Apologies for the delay in responding, as I was researching your case.  As a starting point in investigating your case, may I ask these questions please:

-  Are you using ROS?
-  -  Does your Python script set up more than one pipeline?  I ask this because there was a past Python case where **Failed to set power state** occurred because of this.

https://github.com/IntelRealSense/librealsense/issues/6120#issuecomment-704910981

MartyG-RealSense, 2020-12-09:
Hi @Fred3D-tech   Do you require further assistance with this case, please?  Thanks!

Fred3D-tech, 2020-12-11:
yes.. it is just not working. I just made more tests and opened issue #7976

MartyG-RealSense, 2020-12-12:
Thanks, I have commented at https://github.com/IntelRealSense/librealsense/issues/7976#issuecomment-743738267

MartyG-RealSense, 2020-12-19:
As https://github.com/IntelRealSense/librealsense/issues/7976 is open, I will close this particular case so that https://github.com/IntelRealSense/librealsense/issues/7976 can be focused on.

closed 2020-12-19 by MartyG-RealSense