realsenseai/librealsense #7900: Get wrong cloud Data

shiguangzqz, 2020-12-02:
[image]

The left is the right Cloud Data ，and I always get this data. However I always get the error data which show like picture right when i do a test that run my progress all day and the camera move with my robot. When I get data from the camera dozens of times, I get wrong data like the pictures right once or twice

I run the camera in X86_16.04_ubuntu
the software version is 2.35 
the firmware version is 5.12.5.0

shiguangzqz, 2020-12-02:
can you see the picture？
[image]

shiguangzqz, 2020-12-02:
Also,I check the tempeature when i get the wrong cloud data ,it always in 28℃。And I was wondering if it had anything to do with my mobile camera,but I often get the right data .So,how can i do to make sure i Can get the right data ?Is this camera industrial grade

MartyG-RealSense, 2020-12-02:
Hi @shiguangzqz   May I confirm that when you say the 'right' image, you mean the correct image?  This is important, to make sure that we are not talking about the right-side IR sensor. 

It sounds as though you are doing timed captures during the day instead of recording constantly.  Are you capturing the first frame after starting the pipeline and then stopping the pipeline until the time of the next capture, please?

When auto-exposure is enabled, the camera can take up to several frames to settle down.  So if you are only capturing the first frame, the auto-exposure may not be ready yet and may capture less than ideal images during certain lighting conditions during the day.

This can be avoided if you set a manual exposure value instead of using auto-exposure, but I would imagine that you would want to have auto-exposure enabled if you are capturing during the day to ensure that the camera can adjust automatically to changing environmental conditions.   

If you are starting the pipeline, capturing and then stopping the pipeline with auto-exposure enabled, you could add programming to skip past the first several frames so that by the time that the first capture takes place, the auto-exposure is ready.  Details of how to do this can be found in the link below,

https://github.com/IntelRealSense/librealsense/issues/2269

In regard to your question about whether RealSense is an industrial grade camera: it is widely used for industrial and commercial applications and was designed to be suited for many kinds of applications, including ones with environments that require robust performance in difficult / hazardous environments.  The Intel RealSense cameras do not have IP66 rating for total waterproofing and dust resistance though.

You can improve stability further though, especially in multi-day long run sessions of the camera, by pairing the camera with non-RealSense industrial grade components such as computers and USB equipment.

shiguangzqz, 2020-12-02:
If i start the pipeline ,I won't stop it before turning off the camera.
The "left" is the correct pictures .
So why is the point cloud data on the right generated?Could it be caused by taking pictures immediately after the camera moves to a certain position?
And when I find I get the point cloud data like right ,I try to close the camera and then  reopen the camera,but I still get the wrong result occasionally next time.

MartyG-RealSense, 2020-12-02:
Are you using the camera outdoors, please, where it may have the sun directly in its field of view at some point during the daytime?

shiguangzqz, 2020-12-02:
No，I use the camera indoors

MartyG-RealSense, 2020-12-02:
Okay, thank you very much.  Unless the camera is moving faster than human walking pace, moving the camera should not disrupt the image, unless moving the camera takes a section of the observed surface beyond the maximum depth sensing range of the camera.   I mention movement speed because the D415 model has slower shutters than the D435 and D455 models and so the D415 is less suitable for use with fast motion.

shiguangzqz, 2020-12-02:
I use the D435.
So How Can avoid to get the wrong data like right pictures？

MartyG-RealSense, 2020-12-02:
Is the image being captured automatically or do you push a button to take the capture?

Thank you for your patience with all my questions.   They help to diagnose the problem and work out what the possible cause may be.

MartyG-RealSense, 2020-12-02:
As the problem can sometimes persist if you stop the pipeline and re-start it, I would consider the possibility of it being a USB performance drop that a pipeline stop-start may not correct.  When the problem does occur, a reset of the camera by unplugging it from the USB port and plugging it back in may test whether this is the case.

shiguangzqz, 2020-12-02:
The image being captured automatically.
Our products are being tested for durability,so we are not allowed to re plug the camera .
Does anyone else have a similar problem？

MartyG-RealSense, 2020-12-02:
USB performance drops can occur.  It may happen for non-RealSense related causes, such as a problem on the computer hardware or its operating system, or with the USB (for example, the USB cable or a glitch on a USB hub).  This is why industrial grade computing equipment can help in critical applications.

If you cannot access the USB port, you can program your application with a **hardware_reset()** instruction to reset the camera hardware automatically at a time of your choosing.  This can act like an unplug-replug of the camera.

https://github.com/IntelRealSense/librealsense/issues/2161#issuecomment-408779150

shiguangzqz, 2020-12-02:
OK。I try to use this code .
How long does it take for the device to restart ?

MartyG-RealSense, 2020-12-02:
Dorodnic the RealSense SDK Manager provides advice about the time taken for a hardware_reset in the link below.

https://github.com/IntelRealSense/librealsense/issues/5349#issuecomment-560068750

shiguangzqz, 2020-12-02:
OK Thanks

closed 2020-12-16 by MartyG-RealSense