realsenseai/librealsense #7904: Question about RealSense backend / software stack

jurc192, 2020-12-02:
I am trying to understand how RealSense cameras work on Linux, focusing only on pieces of software that are involved during RGB-D data streaming. If I understand correctly, the RealSense API controls the cameras through Video4Linux2 API by default (V4L backend). 
1. Is there any other documentation on software architecture/implementation details, besides [API_arch](https://github.com/IntelRealSense/librealsense/blob/master/doc/api_arch.md) (maybe linux specific)?
2. If I understand correctly, the RealSense controls video streaming trough userspace library `libuvc` when RSUSB_BACKEND is selected. Is the V4L still involved in any way when using this backend?
3. With a 1000 mile overview, what is the RealSense API doing under the hood when running a minimal RGB-D frame grabbing example using the pipeline interface?
4. Is my idea of software stacks involved correct (ignoring device enumeration, control, metadata, only focused on video streaming functionality)?

V4L BACKEND:
```
        +---------------+
        | RealSense API |
        +---------------+
        |   V4L2 API    |       user space
+----------------------------------------------+
        |     V4L       |       kernel space
        +---------------+
        |    uvcvideo   |
        +---------------+
        |    usbcore    |
+-------+---------------+----------------------+
             USB HC             hardware

```

RSUSB (libuvc) BACKEND:
```
        +---------------+
        | RealSense API |
        +---------------+
        |    libuvc     |
        +---------------+
        |    libusb     |       user space
+-------+---------------+----------------------+
                                kernel space

        +---------------+
        |    usbcore    |
+-------+---------------+----------------------+
             USB HC             hardware

```

Thanks for your great work and support!

dorodnic, 2020-12-08:
Hi @jurc192 

> Is there any other documentation on software architecture/implementation details, besides API_arch (maybe linux specific)?

Not a lot, these are ultimately implementation details. That said, we are happy to share any information about the implementation. 

> If I understand correctly, the RealSense controls video streaming trough userspace library libuvc when RSUSB_BACKEND is selected. Is the V4L still involved in any way when using this backend?

One of the first things RSUSB_BACKEND does is detaching the native driver. From there, V4L2 is no longer involved in any way.
RSUSB is based on the original libuvc from 3 years ago (github.com/ktossell/libuvc) as we state in the attribution list, but it's by now a complete C++ re-write of the library, including HID protocol for IMU devices, so it has little in common with current libuvc master. 

> With a 1000 mile overview, what is the RealSense API doing under the hood when running a minimal RGB-D frame grabbing example using the pipeline interface?

When you are doing:
```cpp
pipeline p;
p.start();
p.wait_for_frames();
```
It more or less corresponds to the following pseudocode -
```cpp
context ctx;
dev = ctx.query_devices().first()
depth_sensor = dev.query_sensors().where(profile.stream() == RS2_STREAM_DEPTH);
color_sensor = dev.query_sensors().where(profile.stream() == RS2_STREAM_COLOR);
syncer s;
frame_queue q;
std::map<rs2_stream, frame> last;
s.start([](frameset fs){
   for (auto f : fs) last[f.profile().stream()]=f;
   auto new_fs = last.values(); // pull .second from every item, put them in a frameset
   q.enqueue(new_fs);
});
depth_sensor.open(...)
color_sensor.open(...)
depth_sensor.start(s);
color_sensor.start(s);

fs = q.wait_for_frame().as<frameset>()
```
So, frame is dequed from V4L2 inside dedicated internal processing thread. On that thread, frame goes through some internal processing to reach right requested format, goes into a syncer, if match is found, on the same thread it is converted into a matched frameset, next its extended into a full frameset with older frames in case match is partial, finally frameset is pushed to a queue and pulled from the application thread (or invoked directly into pipeline callback if you start pipeline with a callback)

> Is my idea of software stacks involved correct (ignoring device enumeration, control, metadata, only focused on video streaming functionality)?

Yes

jurc192, 2020-12-08:
Thanks for your time and explanation, it helps me a lot!

closed 2020-12-08 by jurc192