realsenseai/librealsense #7905: Not working on Nvidia Jetson Nano with Ubuntu 18.04 LTS

erkarim85, 2020-12-02:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435i } | 
| Operating System & Version |   Linux (Ubuntu 18.04 LTS)   | 
| Platform                                 | NVIDIA Jetson |
| Language                            |  {python}                          | 

### Issue Description
Hello,

I am trying to use the camera (D435i) on Jetson Nano using Python3 on Jupyter Notebook.
In order to do that, I needed to import pyrealsense2 as rs
However, I got a message saying it is not defined, so I tried pip install pyrealsense2. However, seems my Jetson Nano is not supported. 
As a work around, I tried to build the source on the Jetson Nano using the terminal commands you provided. 
After I went through everything, I still get the same error that it is not defined. 
I tried copying the pyrealsense2.so and librealsense2.so to the same path of my code file, but it didn't work either. 

I wonder if you can help me with my issue. 

Thanks in advance. 
Karim

MartyG-RealSense, 2020-12-02:
Hi @erkarim85   Pip install packages cannot be used with devices with **Arm** processors such as Jetson and Raspberry Pi, as they are designed for x86 / x64 processors.  There is a method for installing librealsense and Python that has worked consistently for Jetson Nano users though.  The instructions are in the link below.

https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-707501049

If you are not using **Python 3.6**, change the two occurrences of '3.6' in the instructions to match the version number of Python 3 that you are using.

erkarim85, 2020-12-02:
Great! Thank you very much @MartyG-RealSense. I will follow those instructions and try it and let you know.

erkarim85, 2020-12-03:
Thanks again for your help @MartyG-RealSense. 
I can now import pyrealsense2 without any errors. 
However, now I am having another problem, I wanted to test it using the simple test code provided, but I got this error:

`AttributeError: module 'pyrealsense2' has no attribute 'pipeline'`

Any advice?

erkarim85, 2020-12-03:
I tried to type:
`dir(rs)`
and I got:
`['__doc__', '__loader__', '__name__', '__package__', '__path__', '__spec__']`

MartyG-RealSense, 2020-12-03:
There was a Jetson case involving **AttributeError: module 'pyrealsense2' has no attribute 'pipeline'** where the solution in that case was found to be to copy **.so** files to the same directory that the user's project script was in.

https://github.com/IntelRealSense/librealsense/issues/7078#issuecomment-672949007

erkarim85, 2020-12-05:
So I tried that and it didn't work.
I tried one other thing, I uploaded cmake from 3.13 (the first link recommendation) to 3.18, and followed the same instructions for installation from the source.
I reach a weird situation: 
I can see the package in the terminal, but I can't see it in Jupyter Notebook.

I checked from terminal for the file for this package and I put it in the ~/.bashrc file, and I copied this file into the same directory of my file used in jupyter. Still I can only see it on terminal but on Jupyter there is no 'pipeline' in rs module. 

Any recommendations?

again, I appreciate your help.

MartyG-RealSense, 2020-12-05:
There is a **Start testing in Jupyter notebook** guide in the librealsense documentation that may be of some help.

https://colab.research.google.com/drive/10YTLAf2i0R80-XX_6-1gTPtOMD0cRIx5

MartyG-RealSense, 2020-12-05:
My research also found a case where someone was receiving the error **DLL load failed while importing pyrealsense2:  The specified module could not be found** when using the command **import pyrealsense2 as rs** in Jupyter notebook.  

In that particular case, they found that they were running a Jupyter notebook in a python 3.7 virtual environment but Jupyter was picking up Python 3.8.  Once they fixed this, their Jupyter notebook worked.

erkarim85, 2020-12-05:
Thank you @MartyG-RealSense for your replies.
The example of using it won't work for me on jupyter yet, as I told you before, I can see the pyrealsense2 only on python terminal but Jupyter does not see the contents of the package. 

I made sure before that I installed and used the package with the proper python version.
Jupyter Notebook is using Python 3.6.9 and I started python on the terminal using the same version. 

I tried writing on terminal:
 `$ python3.6
>>> import pyrealsense2 as rs
>>> dir(rs)`
This provided me with a lot of functions available in this package (around 20 lines on my terminal):
`['BufData', 'STAEControl', 'STAFactor', 'STCensusRadius', 'STColorControl', 'STColorCorrection', 'STDepthControlGroup', 'STDepthTableControl', 'STHdad', 'STRauColorThresholdsControl', 'STRauSupportVectorControl', 'STRsm', 'STSloColorThresholdsControl', 'STSloPenaltyControl', '__doc__', '__file__', '__loader__', '__name__', '__package__', '__spec__', '__version__', 'adjust_2D_point_to_boundary', 'align', 'auto_calibrated_device', 'calibrated_sensor', 'calibration_status', 'calibration_type', 'camera_info', 'color_sensor', 'colorizer', 'composite_frame', 'config', 'context', 'debug_protocol', 'decimation_filter', 'depth_frame', 'depth_huffman_decoder', 'depth_sensor', 'depth_stereo_sensor', 'device', 'device_calibration', 'device_list', 'disparity_frame', 'disparity_transform', 'distortion', 'dsm_params', 'event_information', 'extension', 'extrinsics', 'filter', 'filter_interface', 'firmware_log_message', 'firmware_log_parsed_message', 'firmware_logger', 'fisheye_sensor', 'format', 'frame', 'frame_metadata_value', 'frame_queue', 'frame_source', 'hdr_merge', 'hole_filling_filter', 'intrinsics', 'is_pixel_in_line', 'log', 'log_message', 'log_severity', 'log_to_callback', 'log_to_console', 'log_to_file', 'max_usable_range_sensor', 'motion_device_intrinsic', 'motion_frame', 'motion_sensor', 'motion_stream', 'motion_stream_profile', 'next_pixel_in_line', 'notification', 'notification_category', 'option', 'option_range', 'options', 'pipeline', 'pipeline_profile', 'pipeline_wrapper', 'playback', 'playback_status', 'pointcloud', 'points', 'pose', 'pose_frame', 'pose_sensor', 'pose_stream', 'pose_stream_profile', 'processing_block', 'quaternion', 'recorder', 'region_of_interest', 'roi_sensor', 'rs2_deproject_pixel_to_point', 'rs2_fov', 'rs2_project_color_pixel_to_depth_pixel', 'rs2_project_point_to_pixel', 'rs2_transform_point_to_point', 'rs400_advanced_mode', 'save_single_frameset', 'save_to_ply', 'sensor', 'sequence_id_filter', 'software_device', 'software_motion_frame', 'software_notification', 'software_pose_frame', 'software_sensor', 'software_video_frame', 'spatial_filter', 'stream', 'stream_profile', 'syncer', 'temporal_filter', 'terminal_parser', 'texture_coordinate', 'threshold_filter', 'timestamp_domain', 'tm2', 'units_transform', 'updatable', 'update_device', 'vector', 'vertex', 'video_frame', 'video_stream', 'video_stream_profile', 'wheel_odometer', 'yuy_decoder', 'zero_order_invalidation']
`

I tried on jupyter to write same code:
` import pyrealsense2 as rs
dir(rs)`

I only got the following line as output:
`['__doc__', '__loader__', '__name__', '__package__', '__path__', '__spec__']`

So, pipeline is not available in the package on the jupyter notebook. It is available on the terminal.

By the way, I tried to copy the *.so files to the same directory of my *.ipynb but it didn't fix it. 

I appreciate if you have another advice...
Thanks again!

erkarim85, 2020-12-05:
Another update:

This problem happens when I access the jupyter notebook by typing in the browser 192.xxx.xx.x:8888 or trying to access jupyter by typing this address on my computer connected to the Jetson nano.

However, I don't have this problem when I use the jetson nano as a computer and type in terminal `jupyter lab`.

Unfortunately, for my application, I need to use it in the headless mode.

MartyG-RealSense, 2020-12-05:
Is the pyrealsense2 installation on the server and not on the local machine that you are using the browser address on?

The link below describes the setup of a headless Jupyter Notebook system where most of the data is on the server and its graphical GUI elements are on the local ms nine.

https://ljvmiranda921.github.io/notebook/2018/01/31/running-a-jupyter-notebook/

MartyG-RealSense, 2020-12-12:
Hi @erkarim85   Do you require further assistance with this case, please?   Thanks!

erkarim85, 2020-12-12:
I still need a little bit more time. It didn’t work that way yet so I am trying to use it in a different way now. 

Best Regards,

Karim H. Erian

MartyG-RealSense, 2020-12-19:
Adding a note to keep this case open for a further time period.

MartyG-RealSense, 2021-01-06:
Hi @erkarim85   Do you have an update that you can provide us, please?  Thanks!

MartyG-RealSense, 2021-01-13:
Case closed due to no further comments received.

closed 2021-01-13 by MartyG-RealSense

erkarim85, 2021-02-18:
@MartyG-RealSense 
Thank you for your help. The Camera is working now with the Jetson Nano. 

I have another problem with the camera. I am trying to set the frame size to 640x480 but it is not working:

`import pyrealsense2 as rs
...
config = rs.config()
config.enable_stream(rs.stream.color,640,480,rs.format.bgr8,15)`

I am trying to use this config function, but I am getting frames of 1280x720 instead. 
Any idea how to fix that?

MartyG-RealSense, 2021-02-18:
Thanks very much @erkarim85 for the update - great news that you got it working!