realsenseai/librealsense #7906: L515/Comvert depth to point cloud

anhTuanBrycen, 2020-12-03:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  L500  | 
| Firmware Version                   | 01.05.00.00 | 
| Operating System & Version |   Win 8.1/10 | 
| Kernel Version (Linux Only)    |                                        | 
| Platform                                 | PC  |
| SDK Version                            |      01.05.00.00                      | 
| Language                               |  C/C++/C#                           | 
| Segment			        |                       | 

### Issue Description
Hi everybody.
I have an L515 camera device in a remote place, through the intenet I received the depth data (required only to receive depth data), now I want to change the depth to point cloud.
###Describe your issue 
1) https://github.com/IntelRealSense/librealsense/tree/master/examples/software-device
2) https://github.com/IntelRealSense/librealsense/blob/master/include/librealsense2/rsutil.h#L49-L69
=> The above two methods only apply to D415 and D435I cameras, but it is not true on L515.
Please give me a solution to solve the above problem, thank you!
###question
1) Is there any method in the 2 above methods that can satisfy my requirement above (Comvert depth to point cloud with camera L515)?
2) Can you provide me the formula to convert depth to point cloud for L515 camera?
3) Or do I wish in the API to have more depth to point cloud conversion even without the L515 camera device connected?
I am looking forward to an answer from the Realsense team. 
Thank you everyone

kkh3049, 2020-12-08:
Hey @anhTuan0712, if you have access to the depth data as a DepthFrame, in C# you can do the following:
```C#
using Intel.RealSense;
//...
//depthFrame is your depth data as a DepthFrame
using (var points = new Intel.RealSense.PointCloud().Process(depthFrame).As<Points>())
{
    var numPoints = points.Count;
    var pointList = new System.Numerics.Vector3[numPoints];
    points.CopyVertices(pointList);
    // Do what you want with the points/store them somewhere else
}
```
That should convert the depth frame into 3d points in meters, I believe.

anhTuan0712, 2020-12-17:
> Hey @anhTuan0712, if you have access to the depth data as a DepthFrame, in C# you can do the following:
> 
> ```cs
> using Intel.RealSense;
> //...
> //depthFrame is your depth data as a DepthFrame
> using (var points = new Intel.RealSense.PointCloud().Process(depthFrame).As<Points>())
> {
>     var numPoints = points.Count;
>     var pointList = new System.Numerics.Vector3[numPoints];
>     points.CopyVertices(pointList);
>     // Do what you want with the points/store them somewhere else
> }
> ```
> 
> That should convert the depth frame into 3d points in meters, I believe.
> (And just so you know, the SDK version and Firmware version are not the same. The current version is 2.40.0.)

Thanks for your help, I just have value Depth want to convert to point Cloud and not use API. but now my work is completed.

RealSenseSupport, 2020-12-21:
Hi @anhTuan0712 

Were you able to figure out a solution since you mention that your work is complete?

*If we don’t hear from you in 7 days, this issue will be closed.

closed 2021-01-12 by RealSenseSupport