realsenseai/librealsense #7908: Playback with callback not working

lmuffang, 2020-12-03:
Hello, 

I am quite new to librealsense and I am having trouble playing back a recorded bag (recorded with `rs-record`). 

I am reading IMU values and camera frames with a callback as shown in the callback example to have all IMU measurements. My code works perfectly with the camera directly but when I read a bag file and use a playback, I do not get all measurements as sometimes I get 2 images with no IMU measurements between them.

Here is a view of the code I use:
```
    auto callback = [&](const rs2::frame& frame)
    {
        std::lock_guard<std::mutex> lock(mutex);
        if (rs2::frameset fs = frame.as<rs2::frameset>())
        {
            // Handle image
        }
        else if (rs2::motion_frame m_frame = frame.as<rs2::motion_frame>())
        {
            if (m_frame.get_profile().stream_name() == "Gyro")
            {
                rs2_vector data_g = m_frame.get_motion_data();
               //Handle Gyro data
            }
            else if (m_frame.get_profile().stream_name() == "Accel") 
            {
                 //Handle Accel data
            }
        }
    };
    bool run_bag = true;  //code works perfectly with this to false
    auto pipe = std::make_shared<rs2::pipeline>();
    rs2::config cfg;
    if (run_bag){
        cfg.enable_device_from_file("<mybag>.bag", false);
    }
    cfg.enable_stream(RS2_STREAM_FISHEYE, 1);
    cfg.enable_stream(RS2_STREAM_FISHEYE, 2);
    cfg.enable_stream(RS2_STREAM_GYRO);
    cfg.enable_stream(RS2_STREAM_ACCEL);
    // Start streaming through the callback with configuration
    pipe->start(cfg, callback);
    rs2::device device = pipe->get_active_profile().get_device();
    rs2::playback playback = device.as<rs2::playback>();
```

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { T265 } | 
| Firmware Version                   | (0.1.0.279) | 
| Operating System & Version |   {Linux (Ubuntu 20)  | 
| Kernel Version (Linux Only)    |  (5.4.0-56)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.<29>.<0> }                          | 
| Language                            |  {C++}                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          |

07hokage, 2021-01-10:
hey @lmuffang , did you try to enable the pose stream too? other than fisheye stream, im not getting any other streams despite enabling them all

aangerma, 2021-02-28:
Hello @lmuffang , 
https://github.com/IntelRealSense/librealsense/pull/8461
can you please check if this PR solves your IMU drops ?
thanks

closed 2021-04-19 by RealSenseSupport